moveit/moveit2_tutorials #27: Port "IKFast Kinematics Solver" tutorial

bostoncleek, 2021-02-04:
Port tutorial to ROS2.

robosherpa, 2021-04-25:
Hi @bostoncleek,

Is it okay if I take this issue?
If you have any pointers for takers of this issue, I would totally appreciate that too.

Thanks!

vatanaksoytezer, 2021-04-29:
Hi @robosherpa,

Thank you for interest! That is the tutorial we are looking to port. We would be more than happy if you help us porting it.

I don't have any specific pointers, but you can take other ported tutorials as a reference.

Feel free to ask if you have any questions while porting it.

vatanaksoytezer, 2021-04-29:
I just noticed https://github.com/ros-planning/moveit2/blob/main/moveit_kinematics/ikfast_kinematics_plugin/scripts/auto_create_ikfast_moveit_plugin.sh is not ported to moveit2 yet. So you would need to port this as well while working on this tutorial. There may be other unported stuff in ikfast on moveit2, I would suggest making a list of these first and posting it here to better evaluate what needs to be done to port this tutorial.

robosherpa, 2021-04-30:
@vatanaksoytezer Thanks for chiming in.

After some study, I discovered that this ticket is outstanding since there are a lot other blocking tickets in the list of tutorials, especially the early ones following [Moveit Commander Tutorial ](http://moveit2_tutorials.picknik.ai/doc/move_group_interface/move_group_interface_tutorial.html#). So, the order of update seems to go as follows:

- [Python Commander Tutorial](https://github.com/ros-planning/moveit2_tutorials/issues/4)
- [Move Group Python Interface](https://github.com/ros-planning/moveit2_tutorials/issues/3)
- [Moveit Visualization Tutorial](https://github.com/ros-planning/moveit2_tutorials/issues/12)

It appears like Python interfaces have a few pending updates in their dependencies. I am likely seeing only the tip of the iceberg at this point, however.

tylerjw, 2023-12-01:
Not yet ported: https://moveit.picknik.ai/main/doc/examples/ikfast/ikfast_tutorial.html