moveit/moveit2_tutorials #59: Migrate From `panda_moveit_config` to `moveit_resources_panda_moveit_config`

DLu, 2021-03-10:
### Description

As discussed here: https://github.com/ros-planning/moveit2_tutorials/blob/main/package.xml#L40-L43

Blocks a bunch of other tutorials.

sea-bass, 2021-03-10:
I think the bigger blocking change for the tutorials is this:

https://github.com/ros-planning/moveit_resources/issues/40

closed 2021-03-11 by JafarAbdi

130s, 2023-06-09:
> As discussed here: https://github.com/ros-planning/moveit2_tutorials/blob/main/package.xml#L40-L43

For the record I _think_ those lines refer to https://github.com/ros-planning/moveit2_tutorials/blob/6ebdf086c344b26ed38b9dec8bc8a4ead07b4754/package.xml#L40-L43, the commit before https://github.com/ros-planning/moveit2_tutorials/pull/61 was merged.

PR #61 (2021-03-11): Add dependency on moveit_resources_panda_moveit_config. 
### Description

Fixes #59.

### Checklist
- [ ] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] While waiting for someone to review your request, please consider reviewing [another open pull request](https://github.com/ros-planning/moveit/pulls) to support the maintainers

[//]: # "You can expect a response from a maintainer within 7 days. If you haven't heard anything by then, feel free to ping the thread. Thank you!"
commit 94083622 (2021-03-11): Add dependency on moveit_resources_panda_moveit_config. Fixes #59. (#61)