moveit/moveit2_tutorials #68: Quickstart_in_rivz problem

AtariaLee, 2021-04-10:
### Description
im learning moveit2 in ros2-foxy.when i run `ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true` by following [https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.rst](url), an error happend.
I would be very thankful for your help!!

### Your environment
* ROS Distro: |Foxy|
* OS Version: e.g. Ubuntu 20.04
* maybe source?
*
*

### Steps to reproduce
demo and files in [https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.rst](url)

### Expected behaviour
when i run `ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true` by following [https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.rst](url), an error happend.

### Backtrace or Console output

lee@lee-VirtualBox:~/ws_moveit2$ ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true
[INFO] [launch]: All log files can be found below /home/lee/.ros/log/2021-04-10-18-23-32-079928-lee-VirtualBox-17769
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:271> exception=InvalidLaunchFileError('py')>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 53, in get_launch_description_from_any_launch_file
    return loader(launch_file_path)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 68, in get_launch_description_from_python_launch_file
    return getattr(launch_file_module, 'generate_launch_description')()
  File "/home/lee/ws_moveit2/install/moveit2_tutorials/share/moveit2_tutorials/launch/demo.launch.py", line 71, in generate_launch_description
    run_move_group_node = Node(package='moveit_ros_move_group',
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 196, in __init__
    normalized_params = normalize_parameters(parameters)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/normalize_parameters.py", line 181, in normalize_parameters
    normalized_params.append(normalize_parameter_dict(param))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/normalize_parameters.py", line 162, in normalize_parameter_dict
    raise TypeError('Unexpected type for parameter value {}'.format(repr(value)))
**TypeError: Unexpected type for parameter value None**

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 273, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 293, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 125, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_description_source.py", line 53, in _get_launch_description
    return get_launch_description_from_any_launch_file(location)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 56, in get_launch_description_from_any_launch_file
    raise InvalidLaunchFileError(extension, likely_errors=exceptions)
**launch.invalid_launch_file_error.InvalidLaunchFileError: Caught exception when trying to load file of format [py]: Unexpected type for parameter value None**

vatanaksoytezer, 2021-04-13:
@AtariaLee I will send a PR, fixing this. In the mean time you can check out https://github.com/ros-planning/moveit2/tree/main/moveit_demo_nodes/run_move_group to achieve a very similar behaviour.

AtariaLee, 2021-04-21:
I tried the new files,but i have the same problem.

vatanaksoytezer, 2021-04-21:
Are you using the current master branch of moveit2?

Did you try [this](https://github.com/ros-planning/moveit2/tree/main/moveit_demo_nodes/run_move_group)?

The PR is not merged yet so if you want to use the fix, you need to pull my branch of moveit2_tutorials (until it is merged) here: https://github.com/vatanaksoytezer/moveit2_tutorials/tree/fix/move_group

closed 2021-04-21 by MarqRazz

AtariaLee, 2021-04-23:
sorry,i am new learner in ros2.it seems that many demos and files are not ported to moveit2,like python interface demos and  others. @vatanaksoytezer

vatanaksoytezer, 2021-04-23:
No worries @AtariaLee. Most tutorials are not ported yet to ros2. But the ported (and merged) ones should work without any problems for now.

AtariaLee, 2021-04-23:
@vatanaksoytezer .thanks a lot!!!but its still too difficult for me to learn moveit2 in ros2 by following tutorials.
By following tutorials ,so many problems was happend,and im a rookie in ros2 and moveit2,so i dont konw how to fix it. That really bothers me.

MarqRazz, 2021-04-23:
@AtariaLee, I'm sorry you are seeing issues but these tutorials are still a work in progress. If there is a warning at the top like this then it has low chances of working for you.
[image]

Are you sure your workspace is up to date? The launch file you are attempting to run works on my machine.

You can update your workspace by cd'ing into it and running `vcs pull src`

Here are the packages I have in my src folder this is what it looks like when I ask vcs to update it (FYI you will probably not have moveit_task_constructor in yours)
[image]

AtariaLee, 2021-04-24:
@MarqRazz   After i update my files,its worked.But i mean there still a lot of tutorials are not worked for moveit2.
And thanks a lot for everyones help!!!

newcanopies, 2021-04-28:
Not sure if related:

`ros2 launch run_moveit_cpp run_moveit_cpp.launch.py`  LOADS Rviz (image left)

but

`ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true`  EMPTY Rviz (image middle)

[image]

vatanaksoytezer, 2021-04-28:
We've already merged my fork. Rviz should be blank with `rviz_tutorial:=true` flag which is meant for you to follow the quickstart tutorial [here](http://moveit2_tutorials.picknik.ai/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html) and add the rviz plugins yourself. If you want an already filled scene you shoud remove `rviz_tutorial:=true` flag and run as `ros2 launch moveit2_tutorials demo.launch.py`.

Let me know if you still have something unexpected after that.

vatanaksoytezer, 2021-04-28:
But it seems you have trouble with mongo, I will try to recreate it in my local machine and see if I can find anything. Let me know if `ros2 launch moveit2_tutorials demo.launch.py` works as expected.

newcanopies, 2021-04-28:
>  We've already merged my fork. Rviz should be blank with `rviz_tutorial:=true` flag which is meant for you to follow the quickstart tutorial [here](http://moveit2_tutorials.picknik.ai/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html) and add the rviz plugins yourself.

thanks @vatanaksoytezer 
`ros2 launch moveit2_tutorials demo.launch.py`  also runs a blank window and produces this new launch log https://gist.github.com/newcanopies/0a0b0f0774a964279198b41c40705bf9

but `rviz_tutorial:=true` should load Rviz with empty scene ---- not a blank white window?
[image]

vatanaksoytezer, 2021-04-28:
I could not reproduce, but I would have some suspicions if:

1) You have something related with mongo running in your computer
2) You have multiple tutorials running at the same time (this should work, but maybe some conflict that I hadn't realized before happened)

Can you try commenting / deleting https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/quickstart_in_rviz/launch/demo.launch.py#L195-L205 and https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/quickstart_in_rviz/launch/demo.launch.py#L216 build again (colcon build) and run?

vatanaksoytezer, 2021-04-28:
> `rviz_tutorial:=true` should load Rviz with empty scene pictured in tutorial

That's true, yes.

newcanopies, 2021-04-28:
Good news, these two can run simultaneous on WSL2
[image]

> * You have something related with mongo running in your computer

running only
 WSL2/Uubuntu > WindowsTerminal >  moveit2
 Windows > Chrome
 
 doubt it's WSL related since it loads the other two at the same time, unless they are not mongo based?

vatanaksoytezer, 2021-04-28:
`run_move_group` is also mongo based. I don't think it should be related to WSL either. Let me know after you try with removing mongo stuff.

newcanopies, 2021-04-28:
> Can you try commenting / deleting https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/quickstart_in_rviz/launch/demo.launch.py#L195-L205 and https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/quickstart_in_rviz/launch/demo.launch.py#L216 build again (colcon build) and run?

rebuilt + sourced install, still the same `ros2 launch moveit2_tutorials demo.launch.py` 
[image]

launch log without mongo https://gist.github.com/newcanopies/719aa474c6420c2126aa8216c85283a1

vatanaksoytezer, 2021-04-28:
I would love to get this working on your configuration as well, I will discuss it with some of the other moveit2 folks to see if they have any idea.

I am not sure if this is unique to you, I would appreciate if any folks seeing this running the tutorial and see if they can reproduce.

For now you can try with `run_move_group` or maybe you can give it a go to demo + rviz files from the PR at https://github.com/ros-planning/moveit_resources/pull/64 as well.

vatanaksoytezer, 2021-04-28:
If the problem persists feel free to open a new issue in this repo and ping me, then I will try to spare some time to see what is wrong and see if I can create a WSL and reproduce.

newcanopies, 2021-04-28:
> try with `run_move_group` or maybe you can give it a go to demo + rviz files from the PR at [ros-planning/moveit_resources#64](https://github.com/ros-planning/moveit_resources/pull/64) as well.

Awesome, thank you!  

> to get this working on your configuration as well

Np, if you ping me with further updates, I will test them on WSL2 : )

PR #69 (2021-04-21): Fix move group namespace and demo.launch.py
- Ports #62, #63 and #64 to the new ROS2 conventions.
- Fixes #68.
- Includes move_group namespace fixes (see https://github.com/ros-planning/moveit2/pull/420)
- Add pre-commit