moveit/moveit2_tutorials #76: moveit_print_planning_scene_info not working as intended with planning scene ros api tutorial

vatanaksoytezer, 2021-04-20:
moveit_print_planning_scene_info fails if called within the context of planning scene ros api tutorial.

See #74.

AtariaLee, 2021-04-29:
After i update my planning scene ros api files,it and builded by`colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release`.But it still some problem like `file 'planning_scene_ros_api_tutorial.launch.py' was not found in the share directory of package 'moveit2_tutorials' which isat'/home/lee/ws_moveit2/install/moveit2_tutorials/share/moveit2_tutorials' `.
Should i  use `concol build `to build all files?

vatanaksoytezer, 2021-04-29:
Two questions to better understand what is wrong:

1) Did you update your moveit2_tutorials package with `git pull`? This tutorial has been recently merged.

2) Are you calling `colcon build` from the root of your workspace and your workspace is sourced?

Such as:

`cd /home/lee/ws_moveit2/ && colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release && source /home/lee/ws_moveit2/install/setup.bash`

AtariaLee, 2021-04-29:
sorry ,my bad.I forgot  `source /home/lee/ws_moveit2/install/setup.bash`.Thanks for your help!