moveit/moveit2_tutorials #83: run_moveit_cpp demo crash and missing Interactive pointer tool (Foxy)

newcanopies, 2021-04-28:
### Description

`run_moveit_cpp` [demo](https://github.com/ros-planning/moveit2/tree/main/moveit_demo_nodes/run_moveit_cpp) is missing the Interactive Tool and crashed. possibly an Xsrc WSL2 issue?

### Env
* ROS2 Foxy
* Ubuntu 20.04/WSL2
* Binary

### Steps to reproduce

run  `ros2 launch run_moveit_cpp run_moveit_cpp.launch.py`
Use the interface according to the tutorial. 
Interactive tool is in the toolbar but does not add the positioning gimbal to scene

### Backtrace or Console output

```
[rviz2-3] Parsing robot urdf xml string.
[rviz2-3] [INFO] [1619643662.982975700] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0008076 seconds
[rviz2-3] [INFO] [1619643662.983048100] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[rviz2-3] [INFO] [1619643663.691225200] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopping planning scene monitor
[rviz2-3] Parsing robot urdf xml string.
[rviz2-3] [INFO] [1619643663.701741400] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0007252 seconds
[rviz2-3] [INFO] [1619643663.701765600] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[rviz2-3] [WARN] [1619643663.710663600] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
[rviz2-3] [INFO] [1619643663.757818700] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[rviz2-3] [INFO] [1619643663.758408700] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/moveit_cpp/publish_planning_scene'

[ERROR] [rviz2-3]: process has died [pid 18802, exit code -11, cmd '/opt/ros/foxy/lib/rviz2/rviz2 -d /home/user/ws_ROS2/moveit2/install/run_moveit_cpp/share/run_moveit_cpp/launch/run_moveit_cpp.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_xbpi4kvw --params-file /tmp/launch_params_cvdjcc3n'].

^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[ros2_control_node-5] [INFO] [1619643780.230165400] [rclcpp]: signal_handler(signal_value=2)
[run_moveit_cpp-4] [INFO] [1619643780.230225600] [rclcpp]: signal_handler(signal_value=2)
[run_moveit_cpp-4] [INFO] [1619643780.230528600] [moveit.ros_planning_interface.moveit_cpp]: Deleting MoveItCpp
[robot_state_publisher-2] [INFO] [1619643780.230249700] [rclcpp]: signal_handler(signal_value=2)
[static_transform_publisher-1] [INFO] [1619643780.230255400] [rclcpp]: signal_handler(signal_value=2)
[run_moveit_cpp-4] [INFO] [1619643780.234566300] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopped publishing maintained planning scene.
[run_moveit_cpp-4] [INFO] [1619643780.235321800] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopping planning scene monitor
[run_moveit_cpp-4] Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
[run_moveit_cpp-4]          at line 127 in /tmp/binarydeb/ros-foxy-class-loader-2.0.2/src/class_loader.cpp
```
Full log https://gist.github.com/newcanopies/26ad2971c65a6e288a849201b0415466

vatanaksoytezer, 2021-04-28:
It is expected not to have an interactive marker in `run_moveit_cpp` demos. `quickstart` demos are the ones that are meant to be used with interactive markers.

For the crash I could not reproduce and I highly suspect that it is related with WSL. Also, unfortunately log doesn't tell me much. It seems that rviz just died. I'll try to install moveit2 and reproduce this in WSL in my free time this week, to help with the problems. If you are just trying to learn moveit2 I would suggest using docker images from [moveit2 docker hub](https://hub.docker.com/r/moveit/moveit2) if that is not much work for you. It would be much smoother experience for you and quite easy for us to reproduce and solve these problems that way.

newcanopies, 2021-04-29:
re. `run_moveit_cpp`
side note: using the **Reset** button crashed the demo yesterday
Launched from VSCode+X11 seems stable so far. 
I have two Q's on the Cpp demo: isn't the Interactive Marker built into Rviz? Why is it only available for Scene Objects in the Cpp demo?
[image]

re. [moveit2 docker hub](https://hub.docker.com/layers/moveit/moveit2/foxy-ci/images/sha256-1df1c841be27f394c9915444b79acf07707cc023e9e0b0df793cfcb0ee2f86cf?context=explore)
I wasn't able to pull $ `docker pull moveit/moveit2` 
```
Using default tag: latest
Error response from daemon: manifest for moveit/moveit2:latest not found: manifest unknown: manifest unknown
```
so I am downloading `docker pull moveit/moveit2:foxy-ci` to test. 

What I'd like to do is adapt either the cpp or python interface tutorials, for use with the URDF/meshes of my custom 6DOF robot to validate controlling it with PoseStamped via move_base_simple/goal in Gazebo. My URDF is so far validated with JointStates in Gazebo. 

Is the [SetupAssistant](http://moveit2_tutorials.picknik.ai/doc/setup_assistant/setup_assistant_tutorial.html) ported yet for configuring custom robots?

Do I need moveit2 in conjunction with with libraries and nodes from [kdl_parser](http://wiki.ros.org/kdl_parser/Tutorials/Start%20using%20the%20KDL%20parser) package to control the EE via PoseStamped in Gazebo?

Thanks for your help!

newcanopies, 2021-04-29:
and getting flicker in the `ros2 launch run_moveit_cpp run_moveit_cpp_fake_controller.launch.py`

[image]

Launch log (if you notice jumps in timestamps, I deleted a bunch of redundant lines for visibility)
https://gist.github.com/newcanopies/5420065d4f741d2ac4decca94fba14d8

newcanopies, 2021-04-30:
last note regarding getting the tutorials to display smooth on WSL2+X11:

This package works like a charm on WSL2 for reference: https://github.com/PickNikRobotics/Universal_Robots_ROS2_Driver
Rviz is launched instantly at full screen and does not crash. 

I bring this up, because `moveit2_tutorials` launch Rviz with a delay, `below the screen` - you have to drag it up by the Nav bar.
And the first tutorial launches a blank even without mongo. 

Thanks for the work on porting moveit to ros2!

vatanaksoytezer, 2021-04-30:
Sorry took some time to answer, didn't have much time yesterday.

> * isn't the Interactive Marker tool built into Rviz and available in any Rviz instance?

You need to upload the parameters and correct the namespaces. See the differences between launch files.
> * why is it only active for Scene Objects in the Cpp demo and not for positioning the `move_group`?

Same as above it is just about the parameters.
> * why don't StoredStates, Plan/Plan&Execute and LoopAnimation work in Cpp demo?

Due to parameters planning libraries for move_group is not uploaded in moveit_cpp tutorial, that's why move_group won't execute these commands.

> * **re.** [moveit2 docker hub](https://hub.docker.com/layers/moveit/moveit2/foxy-ci/images/sha256-1df1c841be27f394c9915444b79acf07707cc023e9e0b0df793cfcb0ee2f86cf?context=explore)
>   I wasn't able to [...]

We don't have a latest tag in moveit2, you can either use `moveit/moveit2:foxy-ci` and build everything from source or `moveit/moveit2:foxy-source` for a docker with source already pulled.

> The [SetupAssistant](http://moveit2_tutorials.picknik.ai/doc/setup_assistant/setup_assistant_tutorial.html) is not ported yet, so what would be the next best method for configuring a custom robot with `moveit2`?

You can use the ros1's setup assitant and modify the launch files you need, I think that would be the easiest way. @JafarAbdi has some experience with it so maybe he can give a more detailed answer on that. 

> Do I need moveit2 in conjunction with libraries & nodes from [kdl_parser](https://github.com/ros/kdl_parser) to control the EE via PoseStamped in Gazebo, or the KDL package provides full kinematics control without need for moveit config?

I think using kdl with moveit would be the easier way to go for your objective.

newcanopies, 2021-05-01:
Thanks very much for the info and workarounds! I was able to get a quick-start with this branch https://github.com/PickNikRobotics/Universal_Robots_ROS2_Driver  <-- the implementation on WSL2 runs super smooth & stable. 

As I poke around the package, I'll circle back with any code to improve the WSL2 instances of RViz.
Other than the flicker, the cpp demo Rviz loads _below the screen space_ . Barely few pixels are visible to click-&-drag. Those who have the TaskBar at bottom may even miss that it loaded. In this screenshot I already nudged-it up to enter the workspace. 
[image]

Also feel free to close the Issue! Many thanks

newcanopies, 2021-05-01:
Thanks very much for great information and workarounds! I was able to get a ROS2 quick-start with this branch https://github.com/PickNikRobotics/Universal_Robots_ROS2_Driver  <-- the implementation on WSL2 runs super smooth & stable. (if it can be a code base reference)

As I poke around the package, I'll circle back with any code to improve the WSL2 instances of RViz.
Other than the flicker, it's curious that the `cpp demo` loads Rviz _below the screen space_ . Barely few pixels are visible to click-&-drag. Those who have the TaskBar at bottom may even miss that it loaded. To make the screenshot, I already dragged it up to enter the workspace. 
[image]

Closing the issue based on clarifications of `cpp demo` as it does run on WSL2. Many thanks 🙌

closed 2021-05-01 by newcanopies