moveit/moveit2_tutorials #106: Race condition in rviz parameters for robot_description

tylerjw, 2021-08-13:
### Description

I think we found this and there is some fix for this but when I run the first tutorial I get this error printed some times:

```
[rviz2-1] [ERROR] [1628822069.664102077] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': parameter 'robot_description' has already been declared
```

### Your environment
* ROS Distro: Foxy
* OS Version: e.g. Ubuntu 20.04
* Binary 2.2.1

### Steps to reproduce
Build main branch of tutorials against foxy and run first tutorial:

```
ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true
```

### Expected behaviour
It shouldn't error.  Another odd behavior I notice is that if I save the rviz config and rerun the demo it starts up like this:

[image]

To clear the Planning Scene Warning I can change the robot_description to something else and change it back and that fixes it and it now looks like this:

[image]
[image]

JafarAbdi, 2021-08-13:
This's a duplicate of https://github.com/ros-planning/moveit2/issues/474, see https://github.com/ros-planning/moveit2/pull/525 for a temporary solution

tylerjw, 2021-08-13:
Generally, though I wonder if we should either use some sort of parameter server (ala ROS 1) for the robot_description or if there should be some sort of robot_description node that various things can request the description through a service or subscribe to a topic where it is published if it changes (latching)?  Thoughts?