moveit/moveit2_tutorials #108: Panda robot model not displaying in rviz2

dpiet, 2021-08-30:
### Description

Similar to issue #68 . Panda robot description model appears to load properly but the model isn't displaying in rviz2.

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04 via WSL2
* Source or Binary build? Binary MoveIt release, Source moveit2_tutorials
* If binary, which release version? MoveIt Foxy release
* If source, which git commit or tag? moveit2_tutorials commit 59cf8c

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

Environment is WSL2 with Win 11 preview build allowing GUI and GPU usage.

`$ ros2 launch run_moveit_cpp run_moveit_cpp.launch.py`
or
`$ ros2 launch moveit2_tutorials demo.launch.py`

### Expected behaviour
Tell us what should happen

Panda robot model mesh displays in rviz2 scene.

### Backtrace or Console output

[Console output](https://gist.github.com/dpiet/5dc59cdddf2cc9694563dea6d27eb403)
[image]

vatanaksoytezer, 2021-08-30:
Hello @dpiet, thanks for reporting the issue!

Looking at the log I am seeing:

```
[ERROR] [ros2 run controller_manager spawner.py panda_hand_controller-7]: process has died [pid 474, exit code 1, cmd 'ros2 run controller_manager spawner.py panda_hand_controller'].
[ros2_control_node-5] [INFO] [1630334922.948028639] [controller_manager]: Loading controller 'joint_state_controller'
[ros2_control_node-5] [ERROR] [1630334922.948097035] [controller_manager]: Available classes:
[ros2_control_node-5] [ERROR] [1630334922.948106816] [controller_manager]: controller_manager/test_controller
[ros2_control_node-5] [ERROR] [1630334922.948111891] [controller_manager]: controller_manager/test_controller_failed_init
[ros2_control_node-5] [ERROR] [1630334922.948116437] [controller_manager]: controller_manager/test_controller_with_interfaces
[ros2_control_node-5] [ERROR] [1630334922.948121251] [controller_manager]: Loader for controller 'joint_state_controller' not found.
[ros2_control_node-5] [INFO] [1630334922.952385891] [controller_manager]: Loading controller 'panda_arm_controller'
[ros2_control_node-5] [ERROR] [1630334922.952437098] [controller_manager]: Available classes:
[ros2_control_node-5] [ERROR] [1630334922.952449073] [controller_manager]: controller_manager/test_controller
[ros2_control_node-5] [ERROR] [1630334922.952454184] [controller_manager]: controller_manager/test_controller_failed_init
[ros2_control_node-5] [ERROR] [1630334922.952458979] [controller_manager]: controller_manager/test_controller_with_interfaces
[ros2_control_node-5] [ERROR] [1630334922.952464177] [controller_manager]: Loader for controller 'panda_arm_controller' not found.
[ERROR] [ros2 run controller_manager spawner.py joint_state_controller-8]: process has died [pid 505, exit code 1, cmd 'ros2 run controller_manager spawner.py joint_state_controller'].
[ERROR] [ros2 run controller_manager spawner.py panda_arm_controller-6]: process has died [pid 471, exit code 1, cmd 'ros2 run controller_manager spawner.py panda_arm_controller'].
```

starting in Line 42. This seems like `controller_manager` was not able to start correctly. Can you make sure ros2_control and ros2_controllers are installed and sourced?

For eg.

```
apt-get install ros-foxy-ros2-control ros-foxy-ros2-controllers
```

and source your /opt/ros/${ROS_DISTRO}/setup.bash ?

dpiet, 2021-08-30:
Hi @vatanaksoytezer thanks for the quick response. Good catch! I double checked that and installed the missing packages. See revised [console output](https://gist.github.com/dpiet/e1f2ab0e66e6619390a450c280dc2985). It seems that on L100 the controller_manager is throwing an error: https://gist.github.com/dpiet/e1f2ab0e66e6619390a450c280dc2985#file-console-output-after-ros2-control-package-install-check-L100. Any suggestions?

vatanaksoytezer, 2021-08-30:
@dpiet I think this is related with https://github.com/ros-planning/moveit_resources/issues/95. @tylerjw do you happen to resolve this? I have a Foxy docker that has source built version of everything and it seem to work fine.

dpiet, 2021-09-05:
I built everything from source and everything is now working as expected. I think the issue should stay open until a new Foxy release resolves it.

vatanaksoytezer, 2021-09-05:
> I built everything from source and everything is now working as expected. I think the issue should stay open until a new Foxy release resolves it.

That's great to know, let's keep it open for now. We are pushing towards a release in 10 September and hopefully everything will be all good after the release and following the next Foxy sync.