moveit/moveit2_tutorials #114: Can't execute path (ros2_control died)

facontidavide, 2021-09-16:
### Description

When launching the tutorial **moveit2_tutorials demo.launch.py**, I am unable to Execute a plan as describe here:

http://moveit2_tutorials.picknik.ai/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html#executing-trajectories-adjusting-speed

I initially thought that it is related to the fact that the initial position of Panda is in collision (and that may play a role), but apparently there is a deeper problem with ros_control:

[image]

```
[ros2 run controller_manager spawner.py panda_arm_controller-7] [INFO] [1631810383.803703288] [spawner_panda_arm_controller]: Waiting for /controller_manager services
[ros2 run controller_manager spawner.py panda_hand_controller-8] [INFO] [1631810383.805819130] [spawner_panda_hand_controller]: Waiting for /controller_manager services
[ros2_control_node-5] [ERROR] [1631810383.820145184] [controller_manager]: The 'type' param not defined for 'joint_state_broadcaster'.
[ERROR] [ros2 run controller_manager spawner.py joint_state_broadcaster-9]: process has died [pid 12322, exit code 1, cmd 'ros2 run controller_manager spawner.py joint_state_broadcaster'].
```

### Environment

* ROS Distro: Foxy
* OS Version: e.g. Ubuntu 20.04
* Binary 

### Steps to reproduce

Run

```
ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true
```

### Expected behaviour

I understand that this should be the initial configuration of the robot:

[image]

### Backtrace or Console output

Full console output: 
https://gist.github.com/facontidavide/4cb81f59be870da054494eb73e23ed7d

facontidavide, 2021-09-16:
I found the problem: it is an outdated version of the package **moveit_resources_panda_moveit_config**.

The latest version in the repository (`ros2` branch) works fine: https://github.com/ros-planning/moveit_resources/tree/ros2/panda_moveit_config

Therefore the solution is to simply release a new version of that package for Foxy :smile:

facontidavide, 2021-09-16:
A quick workaround would be to add the repository https://github.com/ros-planning/moveit_resources to file [moveit2_tutorials.repos](https://github.com/ros-planning/moveit2_tutorials/blob/main/moveit2_tutorials.repos)

JafarAbdi, 2021-09-16:
Fixed by #115

closed 2021-09-16 by JafarAbdi