moveit/moveit2_tutorials #160: Unable to locate package ros-foxy-rviz-visual-tools

YXGuan, 2021-10-31:
### Description

Unable to locate package ros-foxy-rviz-visual-tools

### Your environment
* ROS Distro: Foxy
* OS Version: e.g. Ubuntu 20.04
* Source build?
* If binary, which release version?
* If source, which git commit or tag?

Hi there,

I have run into some issues when following along the Moveit2 Getting Started [tutorial](http://moveit2_tutorials.picknik.ai/doc/getting_started/getting_started.html). Any idea or suggestion will be much appreciated!

The command: `rosdep install -r --from-paths . --ignore-src --rosdistro  foxy -y `

The output: `
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
moveit2_tutorials: Cannot locate rosdep definition for [moveit_visual_tools]
Continuing to install resolvable dependencies...
executing command [sudo -H apt-get install -y ros-foxy-rviz-visual-tools]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-foxy-rviz-visual-tools
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-foxy-rviz-visual-tools] failed
  apt: Failed to detect successful installation of [ros-foxy-rviz-visual-tools]
`

AndyZe, 2021-10-31:
Hmm, it looks like `moveit_visual_tools` was never released for Foxy (see [here](https://github.com/ros/rosdistro/blob/master/foxy/distribution.yaml)) but it was released for Galactic and Rolling (see [here](https://github.com/ros/rosdistro/blob/e1c9f48dc42f43b3d507d7dfd00a13ec839dc98d/rolling/distribution.yaml#L1721)).

Can you please check that you have indeed checked out the Foxy branch of moveit2_tutorials and moveit2

AndyZe, 2021-10-31:
Also, it look like this step should have clone `moveit_visual_tools` into your workspace. Can you please check if it's there?

`vcs import < moveit2_tutorials/moveit2_tutorials.repos`

Note, moveit_visual_tools is definitely included in that `moveit2_tutorials.repos`. (see [here](https://github.com/ros-planning/moveit2_tutorials/blob/2eb74fef6bbdc925341afaa8b6178e1a532a711c/moveit2_tutorials.repos#L10))

YXGuan, 2021-10-31:
Hey Andy,

Thanks for the help. You are right, moveit_visual_tools is installed in the correct workspace. 
<pre><font color="#4E9A06"><b>yuxiang@yuxiang</b></font>:<font color="#3465A4"><b>~/ws_moveit2/src</b></font>$ vcs import &lt; moveit2_tutorials/moveit2_tutorials.repos
...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./moveit_resources</b></font><font color="#3465A4"> (git) ===</font>

Already on &apos;ros2&apos;
Your branch is up to date with &apos;origin/ros2&apos;.
<font color="#3465A4">=== </font><font color="#3465A4"><b>./moveit_visual_tools</b></font><font color="#3465A4"> (git) ===</font>

Already on &apos;ros2&apos;
Your branch is up to date with &apos;origin/ros2&apos;.
<font color="#3465A4">=== </font><font color="#3465A4"><b>./rviz_visual_tools</b></font><font color="#3465A4"> (git) ===</font>

Already on &apos;foxy-devel&apos;
Your branch is up to date with &apos;origin/foxy-devel&apos;.
</pre>

Then, I deleted and re-installed Moveit, here's a new error now:

After entering the command: colcon build --mixin release

It gives me the error output:

Starting >>> moveit_resources_panda_description
Starting >>> moveit_resources_fanuc_description
Starting >>> rviz_visual_tools
Starting >>> moveit_resources_pr2_description
Finished <<< moveit_resources_fanuc_description [1.03s]
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_panda_description [1.06s]
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_pr2_description [1.15s]                 
Finished <<< moveit_resources_fanuc_moveit_config [0.91s]             
Finished <<< moveit_resources_panda_moveit_config [0.92s]
Starting >>> moveit_resources
Finished <<< moveit_resources [0.81s]                                 
--- stderr: rviz_visual_tools                         
CMake Error at CMakeLists.txt:29 (find_package):
  By not providing "Findeigen_stl_containers.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "eigen_stl_containers", but CMake did not find one.

  Could not find a package configuration file provided by
  "eigen_stl_containers" with any of the following names:

    eigen_stl_containersConfig.cmake
    eigen_stl_containers-config.cmake

  Add the installation prefix of "eigen_stl_containers" to CMAKE_PREFIX_PATH
  or set "eigen_stl_containers_DIR" to a directory containing one of the
  above files.  If "eigen_stl_containers" provides a separate development
  package or SDK, be sure it has been installed.

---
Failed   <<< rviz_visual_tools [3.64s, exited with code 1]

Summary: 6 packages finished [3.85s]
  1 package failed: rviz_visual_tools
  1 package had stderr output: rviz_visual_tools
  2 packages not processed

tylerjw, 2021-11-01:
That package comes from the rosdep install step.  After you added rviz_visual_tools to your workspace you need to run this again:

```
rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
```

ichdich, 2022-03-16:
apt-cache search ros-foxy-moveit-visual-tools returns none.
moveit_visual_tools package can be downloaded from https://github.com/ros-planning/moveit_visual_tools .
moveit_visual_tools has branch name "foxy".
I downloaded it and successfully build moveit2_tutorials package.

YXGuan, 2022-07-22:
Thanks a lot for all your help!

closed 2022-07-22 by YXGuan