moveit/moveit2_tutorials #168: panda_arm.xacro links to melodic source

tylerjw, 2021-11-01:
### Description

On this page: http://moveit2_tutorials.picknik.ai/doc/moveit_cpp/moveitcpp_tutorial.html

There is a link to panda_arm.xacro in a melodic branch.  Should there be a link to the foxy version of this?

rhaschke, 2021-11-02:
We don't have specific ROS2 branches of `panda_moveit_config`, do we? We are currently in the process of finalizing the Noetic branch, which should become the default for Melodic and Noetic.

vatanaksoytezer, 2021-11-02:
> We don't have specific ROS2 branches of `panda_moveit_config`, do we? We are currently in the process of finalizing the Noetic branch, which should become the default for Melodic and Noetic.

Not for panda spesifically but we do have a `ros2` branch of `moveit_resources` which includes `panda_moveit_config`: https://github.com/ros-planning/moveit_resources/tree/ros2. That needs a ton of cleaning and a new set of config generation when MSA is ready in ROS2 though.

closed 2021-11-03 by tylerjw