moveit/moveit2_tutorials #173: Issue with moveit_cpp_tutorial.launch.py

ZhongBaode, 2021-11-02:
### Description

Hi, I was doing the moveit tutorials and when i came to "MoveitCpp Tutorial" section, I realized i could not run the launch file.

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04
* Source or Binary build? Source
* If binary, which release version?
* If source, which git commit or tag? Foxy branch

### Steps to reproduce
Run the following command
```
ros2 launch moveit2_tutorials moveit_cpp_tutorial.launch.py
```

### Expected behaviour
Rviz 2 is suppose to run but doesn't.

### Backtrace or Console output

The error that comes out in console:
```
file 'moveit2_cpp_tutorial.launch.py' was not found in the share directory of package 'moveit2_tutorials' which is at '/home/rosi/ws_moveit2/install/moveit2_tutorials/share/moveit2_tutorials'
```

tylerjw, 2021-11-02:
This should have been fixed by this: https://github.com/ros-planning/moveit2_tutorials/pull/169

Do you mind pulling the repo again, rebuilding, and trying again?

ZhongBaode, 2021-11-02:
Okay! thanks it works now

closed 2021-11-02 by ZhongBaode