moveit/moveit2_tutorials #185: Joint Limits are not included in tutorial launch files

vatanaksoytezer, 2021-11-03:
Basically what the title says. We should consider joint limits read from the config file of the robots in the tutorials.

sea-bass, 2023-03-25:
Bumping this issue, because I was just testing something on the latest moveit2 branch and one of our most basic examples:

```
ros2 launch moveit_resources_panda_moveit_config demo.launch.py
```

has errors about required parameters for the panda config.

```
[rviz2-1] [ERROR] [1679761789.691008064] [moveit_trajectory_processing.time_optimal_trajectory_generation]: No acceleration limit was defined for joint panda_joint1! You have to define acceleration limits in the URDF or joint_limits.yaml
```

AndyZe, 2023-03-25:
I actually don't see that error?! I have the `main` or `ros2` branches of moveit2, moveit2_tutorials, and moveit_resources.

AndyZe, 2023-03-25:
Closed by https://github.com/ros-planning/moveit_resources/pull/165

closed 2023-03-25 by AndyZe

sea-bass, 2023-03-25:
Sorry, the issue was in moveit_resources, but not this repo. I forgot to amend this. Thanks, Andy!