moveit/moveit2_tutorials #188: Graph_msgs missing from Foxy tutorials

stephanie-eng, 2021-11-03:
### Description
When running this command:
```rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y```

Got the following error:

```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
moveit2_tutorials: Cannot locate rosdep definition for [gripper_controllers]
moveit_visual_tools: Cannot locate rosdep definition for [graph_msgs]
```

vatanaksoytezer, 2021-11-03:
Thanks a lot for the issue and the fix @stephanie-eng!

Though, it seems like I can both install graph_msgs and gripper controllers via apt-get:

```
sudo apt-get install ros-foxy-gripper-controllers
sudo apt-get install ros-foxy-graph-msgs
```

Can you try running and sudo apt-get update and trying again?

Can you also confirm you can run the same commands?

vatanaksoytezer, 2021-11-03:
@stephanie-eng let me know if we should close the issue.

stephanie-eng, 2021-11-08:
This issue still exists in the main branch as there is no package for ros-galactic-graph-msgs

vatanaksoytezer, 2021-11-08:
Seems like graph_msgs is already released though: https://github.com/ros/rosdistro/pull/30780 and CI could install it sucesfully: https://github.com/ros-planning/moveit2_tutorials/runs/4094661154?check_suite_focus=true#step:6:195. I'll check this out.

stephanie-eng, 2021-11-11:
After the latest sync, this issue is now resolved.

closed 2021-11-11 by stephanie-eng