moveit/moveit2_tutorials #192: tab is a confusing word

tylerjw, 2021-11-03:
### Description

On this page there is this block of text:
https://github.com/ros-planning/moveit2_tutorials/blob/foxy/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.rst

```
The planning scene robot using the Show Robot Visual checkbox in the Scene Robot tab.
The planned path using the Show Robot Visual checkbox in the Planned Path tab.
The start state using the Query Start State checkbox in the Planning Request tab.
The goal state using the Query Goal State checkbox in the Planning Request tab.
```

The word `tab` seems wrong as these are not tabs (and there are other things that are tabs in the UI).  We need a better word.

gavanderhoorn, 2021-11-03:
I believe those would be *tree menu*s or *tree view*s.

tylerjw, 2021-11-03:
Thank you @stephanie-eng for fixing this.

closed 2021-11-03 by tylerjw

gavanderhoorn, 2021-11-03:
np