moveit/moveit2_tutorials #193: Rviz crash

stephanie-eng, 2021-11-03:
### Description
Rviz crashed during planning

http://moveit2_tutorials.picknik.ai/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html#plan-cartesian-motions

```
[rviz2-1] [ERROR] [1635957846.579734999] [rviz2]: ItemIdentityException: Material with the name package://moveit_resources_panda_description/meshes/visual/link6.daeMaterial141120Robot already exists. in ResourceManager::add at /tmp/binarydeb/ros-foxy-rviz-ogre-vendor-8.2.2/obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceManager.cpp (line 148)
[rviz2-1] terminate called after throwing an instance of 'Ogre::ItemIdentityException'
[rviz2-1]   what():  ItemIdentityException: Material with the name package://moveit_resources_panda_description/meshes/visual/link6.daeMaterial141120Robot already exists. in ResourceManager::add at /tmp/binarydeb/ros-foxy-rviz-ogre-vendor-8.2.2/obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceManager.cpp (line 148)
[ERROR] [rviz2-1]: process has died [pid 4788, exit code -6, cmd '/opt/ros/foxy/lib/rviz2/rviz2 -d /root/ws_moveit2/install/moveit2_tutorials/share/moveit2_tutorials/launch/panda_moveit_config_demo_empty.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_7djex0yz --params-file /tmp/launch_params_suqbmlcs --params-file /tmp/launch_params_557zeiji --params-file /tmp/launch_params_1nr0g4zo'].
```

corycrean, 2021-12-15:
I had this issue as well.  I was able to reproduce it using the following steps:

1. Launch the tutorial:
`ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true`
2. Adjust the Goal State (the orange robot).
3. Check the "Query Start State" checkbox, and adjust the Start State (the green robot).
4. Check the "Show Trail" checkbox.
5. Click "Plan" under the "Planning" tab.
6. Once the trail appears, uncheck and re-check the "Show Trail" checkbox (sometimes I had to do this twice before RViz would crash).

At @tylerjw's suggestion, I made a video of the process:

https://user-images.githubusercontent.com/96085371/146263379-c487f91d-5376-4f4b-a902-2997c8796c55.mp4

Here is the error message that was generated:

> [rviz2-1] [ERROR] [1639601859.956869061] [rviz2]: ItemIdentityException: Material with the name package://moveit_resources_panda_description/meshes/visual/link6.daeMaterial141120Robot already exists. in ResourceManager::add at /tmp/binarydeb/ros-foxy-rviz-ogre-vendor-8.2.5/obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceManager.cpp (line 148)
[rviz2-1] terminate called after throwing an instance of 'Ogre::ItemIdentityException'
[rviz2-1]   what():  ItemIdentityException: Material with the name package://moveit_resources_panda_description/meshes/visual/link6.daeMaterial141120Robot already exists. in ResourceManager::add at /tmp/binarydeb/ros-foxy-rviz-ogre-vendor-8.2.5/obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceManager.cpp (line 148)
[ERROR] [rviz2-1]: process has died [pid 103592, exit code -6, cmd '/opt/ros/foxy/lib/rviz2/rviz2 -d /home/ccrean/ws_moveit2/install/moveit2_tutorials/share/moveit2_tutorials/launch/panda_moveit_config_demo_empty.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_7jaf8_7n --params-file /tmp/launch_params_4iuqtz8b --params-file /tmp/launch_params_ycytae0e --params-file /tmp/launch_params_wyx6mufs'].

corycrean, 2022-01-18:
This issue has been addressed by this PR: https://github.com/ros2/rviz/pull/811

closed 2022-01-18 by corycrean