moveit/moveit2_tutorials #202: the robot won't move when I following official tutorials[Move Group C++ Interface]

mengyin98, 2021-11-09:
### Description

I am studying the official tutorials of moveit2, things went well until I get to the chapter [Move Group C++ Interface]
According to the tutorial, I should see the robot arm moving from one pose to another or "climb" over the obstacle with a cylinder on its gripper. However, I can only see robot arm simulating in step 3(The robot moves its arm back to a new pose goal while maintaining the end-effector level.) and step 4(The robot moves its arm along the desired Cartesian path (a triangle down, right, up+left). In step1 and step2, I can only see the goal pose, no moving. In step6 7, I can only see the obstacle and cylinder and cartesian path, but robot keep still.

All the code  and packages was downloaded following the tutorials[Getting Started].

### Your environment
* ROS Distro: Foxy
* OS Version: e.g. Ubuntu 20.04

Here are some error information when press the "next"
[ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/base/src/Planner.cpp:257 - panda_arm/panda_arm: Skipping invalid start state (invalid state)
[ERROR] [1636452054.051086261] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:207 - panda_arm/panda_arm: Motion planning start tree could not be initialized!

[INFO] [1636452054.051090566] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/src/SimpleSetup.cpp:140 - No solution found after 0.000033 seconds
[WARN] [1636452054.061120905] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/base/goals/src/GoalLazySamples.cpp:129 - Goal sampling thread never did any work.

[INFO] [1636452054.061204578] [moveit.ompl_planning.model_based_planning_context]: Unable to solve the planning problem
[move_group-4]

 [INFO] [1636452054.061237439] [moveit_move_group_default_capabilities.move_action_capability]: No motion plan found

vatanaksoytezer, 2021-11-09:
Hello @mengyin98, thanks for the issue! Do you mind sharing the full log from the start, and possibly a video / gif on what's happening so that we can try to understand what might the problem?

mengyin98, 2021-11-10:
> Hello @mengyin98, thanks for the issue! Do you mind sharing the full log from the start, and possibly a video / gif on what's happening so that we can try to understand what might the problem?

Thanks for your help sir, here are all the log from the shell and the video

[move_group-4] You can start planning now!
[move_group-4] 
[move_group-4] [INFO] [1636509654.205142648] [moveit_move_group_default_capabilities.move_action_capability]: Received request
[move_group-4] [INFO] [1636509654.205393956] [moveit_move_group_default_capabilities.move_action_capability]: executing..
[move_group-4] [INFO] [1636509654.218207394] [moveit_move_group_default_capabilities.move_action_capability]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[move_group-4] [INFO] [1636509654.257353411] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'panda_link5' (type 'Robot link') and 'panda_link7' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[move_group-4] [INFO] [1636509654.257375816] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[move_group-4] [INFO] [1636509654.257380707] [moveit_ros.fix_start_state_collision]: Start state appears to be in collision with respect to group panda_arm
[move_group-4] [WARN] [1636509654.263700543] [moveit_ros.fix_start_state_collision]: Unable to find a valid state nearby the start state (using jiggle fraction of 0.020000 and 100 sampling attempts). Passing the original planning request to the planner.
[move_group-4] [INFO] [1636509654.299678507] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-4] [WARN] [1636509654.342196237] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/base/src/Planner.cpp:257 - panda_arm/panda_arm: Skipping invalid start state (invalid state)
[move_group-4] [ERROR] [1636509654.342354070] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:207 - panda_arm/panda_arm: Motion planning start tree could not be initialized!
[move_group-4] [INFO] [1636509654.342381106] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/src/SimpleSetup.cpp:140 - No solution found after 0.000354 seconds
[move_group-4] [INFO] [1636509654.342511826] [moveit.ompl_planning.model_based_planning_context]: Unable to solve the planning problem
[move_group-4] [INFO] [1636509654.342627062] [moveit_move_group_default_capabilities.move_action_capability]: No motion plan found. No execution attempted.
[move_group-4] [INFO] [1636509656.318558536] [moveit_move_group_default_capabilities.move_action_capability]: Received request
[move_group-4] [INFO] [1636509656.318603160] [moveit_move_group_default_capabilities.move_action_capability]: executing..
[move_group-4] [INFO] [1636509656.419004768] [moveit_move_group_default_capabilities.move_action_capability]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[move_group-4] [INFO] [1636509656.419586387] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'panda_link5' (type 'Robot link') and 'panda_link7' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[move_group-4] [INFO] [1636509656.419646201] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[move_group-4] [INFO] [1636509656.419677308] [moveit_ros.fix_start_state_collision]: Start state appears to be in collision with respect to group panda_arm
[move_group-4] [WARN] [1636509656.433509642] [moveit_ros.fix_start_state_collision]: Unable to find a valid state nearby the start state (using jiggle fraction of 0.020000 and 100 sampling attempts). Passing the original planning request to the planner.
[move_group-4] [INFO] [1636509656.433824356] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-4] [WARN] [1636509656.433892117] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/base/src/Planner.cpp:257 - panda_arm/panda_arm: Skipping invalid start state (invalid state)
[move_group-4] [ERROR] [1636509656.433907251] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:207 - panda_arm/panda_arm: Motion planning start tree could not be initialized!
[move_group-4] [INFO] [1636509656.433911427] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/src/SimpleSetup.cpp:140 - No solution found after 0.000038 seconds
[move_group-4] [WARN] [1636509656.443902022] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/base/goals/src/GoalLazySamples.cpp:129 - Goal sampling thread never did any work.
[move_group-4] [INFO] [1636509656.443992530] [moveit.ompl_planning.model_based_planning_context]: Unable to solve the planning problem
[move_group-4] [INFO] [1636509656.444027444] [moveit_move_group_default_capabilities.move_action_capability]: No motion plan found. No execution attempted.
[move_group-4] [INFO] [1636509658.919347986] [moveit_move_group_default_capabilities.move_action_capability]: Received request
[... cut ...]
[move_group-4] [WARN] [1636509689.567090437] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/base/goals/src/GoalLazySamples.cpp:129 - Goal sampling thread never did any work.
[move_group-4] [INFO] [1636509689.567168646] [moveit.ompl_planning.model_based_planning_context]: Unable to solve the planning problem
[move_group-4] [INFO] [1636509689.567197535] [moveit_move_group_default_capabilities.move_action_capability]: No motion plan found. No execution attempted.

https://user-images.githubusercontent.com/92976030/141036914-1ae1c3cd-fe04-4056-bd21-c912b76598f8.mp4

vatanaksoytezer, 2021-11-10:
Unfortunately I can't reproduce this.

Though from the video I can see that the arm is not positioned correctly but instead starting from its default position. Normally the arm should look like the following in the start position:

[image]

I am guessing something wrong with ros2_control packages but I would need to see the logs before `[move_group-4] You can start planning now!` as well. If the logs are too long for your shell, you can redirect the logs to a file with the following command:

`ros2 launch moveit2_tutorials move_group.launch.py > log.txt`

and send the file so we can have a look.

mengyin98, 2021-11-11:
> Unfortunately I can't reproduce this.
>
> Though from the video I can see that the arm is not positioned correctly but instead starting from its default position. Normally the arm should look like the [...]

Thanks again, I will upload the log.txt

Meanwhile, I use "colcon build --packages-select move_group_interface_tutorial"  in package move_group_interface(~/moveit2/src/moveit2_tutorials/doc) to see if there are some problems,  turns out there are a lot of errors, I look up it in the internet and seems it is because something wrong with DLL. I am not sure. 

Here are the logs when I build this package.
[Uploading log.txt…]()

~/moveit2/src/moveit2_tutorials/doc$ colcon build --packages-select move_group_interface_tutorial
Starting >>> move_group_interface_tutorial
--- stderr: move_group_interface_tutorial                                
/usr/bin/ld: CMakeFiles/move_group_interface_tutorial.dir/src/move_group_interface_tutorial.cpp.o: in function `main':
move_group_interface_tutorial.cpp:(.text+0x41d): undefined reference to `moveit::planning_interface::MoveGroupInterface::MoveGroupInterface(std::shared_ptr<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<tf2_ros::Buffer> const&, rclcpp::Duration const&)'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0x485): undefined reference to `moveit::planning_interface::PlanningSceneInterface::PlanningSceneInterface(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0x4c8): undefined reference to `moveit::planning_interface::MoveGroupInterface::getCurrentState(double) const'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0x515): undefined reference to `moveit::planning_interface::MoveGroupInterface::getRobotModel() const'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0x7ca): undefined reference to `moveit::planning_interface::MoveGroupInterface::getPlanningFrame[abi:cxx11]() const'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0x912): undefined reference to `moveit::planning_interface::MoveGroupInterface::getEndEffectorLink[abi:cxx11]() const'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0xab8): undefined reference to `moveit::planning_interface::MoveGroupInterface::getJointModelGroupNames[abi:cxx11]() const'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0xad2): undefined reference to `moveit::planning_interface::MoveGroupInterface::getJointModelGroupNames[abi:cxx11]() const'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0xbfc): undefined reference to `moveit::planning_interface::MoveGroupInterface::setPoseTarget(geometry_msgs::msg::Pose_<std::allocator<void> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0xc42): undefined reference to `moveit::planning_interface::MoveGroupInterface::plan(moveit::planning_interface::MoveGroupInterface::Plan&)'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0x1099): undefined reference to `moveit::planning_interface::MoveGroupInterface::getCurrentState(double) const'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0x110c): undefined reference to `moveit::planning_interface::MoveGroupInterface::setJointValueTarget(std::vector<double, std::allocator<double> > const&)'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0x1127): undefined reference to `moveit::planning_interface::MoveGroupInterface::setMaxVelocityScalingFactor(double)'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0x1142): undefined reference to `moveit::planning_interface::MoveGroupInterface::setMaxAccelerationScalingFactor(double)'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0x115b): undefined reference to `moveit::planning_interface::MoveGroupInterface::plan(moveit::planning_interface::MoveGroupInterface::Plan&)'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0x14d2): undefined reference to `moveit::planning_interface::MoveGroupInterface::setPathConstraints(moveit_msgs::msg::Constraints_<std::allocator<void> > const&)'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0x14f7): undefined reference to `moveit::planning_interface::MoveGroupInterface::getCurrentState(double) const'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0x1622): undefined reference to `moveit::planning_interface::MoveGroupInterface::setStartState(moveit::core::RobotState const&)'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0x166e): undefined reference to `moveit::planning_interface::MoveGroupInterface::setPoseTarget(geometry_msgs::msg::Pose_<std::allocator<void> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0x16a7): undefined reference to `moveit::planning_interface::MoveGroupInterface::setPlanningTime(double)'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0x16c0): undefined reference to `moveit::planning_interface::MoveGroupInterface::plan(moveit::planning_interface::MoveGroupInterface::Plan&)'
/usr/bin/ld: move_group_interface_tutorial.cpp:(.text+0x1a48): undefined reference to `moveit::planning_interface::MoveGroupInterface::clearPathConstraints()'
[... cut ...]
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/move_group_interface_tutorial.dir/build.make:431：move_group_interface_tutorial] 错误 1
make[1]: *** [CMakeFiles/Makefile2:78：CMakeFiles/move_group_interface_tutorial.dir/all] 错误 2
make: *** [Makefile:141：all] 错误 2
---
Failed   <<< move_group_interface_tutorial [23.0s, exited with code 2]
                                 
Summary: 0 packages finished [23.1s]
  1 package failed: move_group_interface_tutorial
  1 package had stderr output: move_group_interface_tutorial

RoboHack04, 2022-05-04:
@mengyin98 I am not sure if you are still looking for an answer, but for anyone who is facing similar issue.....the reason is with the initial_position xml used in ros2_control. This can be found in "panda.ros2_control.xacro" file of your panda_moveit_config package. 

You have to change the xml format to -
```
<param name="initial_position">${initial_positions['panda_joint1']}</param>
```

Also check this [`link`](https://github.com/ros-controls/ros2_control/issues/622)

closed 2022-05-09 by JafarAbdi