moveit/moveit2_tutorials #206: Realtime Servo - Initial Configuration Collision

mhubii, 2021-11-15:
When following the Realtime Servo tutorial http://moveit2_tutorials.picknik.ai/doc/realtime_servo/realtime_servo_tutorial.html, the robot launches in a zero configuration, see below

[image]

This causes a collision, see below

```shell
[servo_cpp_interface_demo-3] [WARN] [1636972632.402518987] [moveit_collision_detection.collision_common]: Objects in collision (printing 1st of 1 pairs): panda_link5, panda_link7
```

I launched the servo via `ros2 launch moveit2_tutorials servo_cpp_interface_demo.launch.py`.

tylerjw, 2021-11-15:
How have you installed moveit?  Is this on Foxy, Galactic, or Rolling.  Did you build moveit from the source or install a binary?

tylerjw, 2021-11-15:
I ran locally using the foxy branch (the tutorial you linked) with the tutorials built and installed as shown in the Getting Started tutorial and it drifts around until it is in collision and stops as the tutorial should.

[image]

mhubii, 2021-11-15:
thanks for coming back, using Foxy. Moveit is installed via the package manager. The tutorials are compiled from source, as outlined here https://github.com/ros-planning/moveit2_tutorials#moveit2-tutorials-source-build

np, surely it is a stupid mistake. Feel free to close

tylerjw, 2021-11-15:
> thanks for coming back, using Foxy. Moveit is installed via the package manager. The tutorials are compiled from source, as outlined here https://github.com/ros-planning/moveit2_tutorials#moveit2-tutorials-source-build
> 
> np, surely it is a stupid mistake. Feel free to close

I'm just trying to figure out how this happened to you.  I think those instructions are probably out of date as it has you building the main branch of the tutorials.  Do you mind trying with the `foxy` branch of the moveit2_tutorials repo?

mhubii, 2021-11-15:
just trying, I'll let you know in a second

tylerjw, 2021-11-15:
Building from the main branch I can also reproduce the error you are experiencing.  Please try building the `foxy` branch of thetutorials.  The main branch of the tutorials is not fully tested.  Eventually, the main branch will be for rolling (which will become Humble once it exists).

mhubii, 2021-11-15:
yes solves the issue, thanks a lot. Very nice work btw

closed 2021-11-15 by mhubii

tylerjw, 2021-11-15:
Sweet, thank you.  I'm sorry the readme isn't up to date.  I've been doing a bunch of work to try to make the tutorials more bug fee.  If you notice anything else please reach out and post another issue.