moveit/moveit2_tutorials #224: Joint limits from urdf acting as integers

michalina222, 2021-11-23:
### Description

Hi. I tried using this tutorial. However when I run it using 
```
ros2 launch moveit2_tutorials demo.launch.py
```
I cannot get the joints to move to the limits. Example: joint limit for panda_joint1 is 2.9 radians, but in the joints tab in motion planning I can only move it to 115 degrees, which is around 2 radians. Don't know if this is a moveit2 problem or tutorial problem though.
I checked joint limits set in the panda.urdf, didn't change anything. 
### Your environment
* ROS2 Distro: Foxy
* OS Version: Ubuntu 20.04
*  Binary build: Version: 2.2.1-1focal.20211014.210158

### Steps to reproduce
What I said earlier. 
1. Download moveit2 tutorials like on https://moveit.picknik.ai/foxy/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html#getting-started
2. run it with `ros2 launch moveit2_tutorials demo.launch.py`
3. In MotionPlanning section, in the Joints tab try to manually set a joint to its limit: panda_link1 to 166,6 degrees (2.9 rad)
4. it jumps back to 115 degrees (around 2 radians)
### Expected behaviour
It should set joint state normally. 

[image]

michalina222, 2021-11-23:
Forgot: when limits are set to zero point something, iot doesn't move, when they are set to one point something, I can't move it further than one radian

ravnicas, 2022-09-01:
I also experienced this problem for a while.

The solution was to change the locale, as described in the moveit2 from source install description. I'm not sure whether this applies for binaries as well.

closed 2023-12-01 by tylerjw