moveit/moveit2_tutorials #240: Cannot switch to MoveIt 1 tutorials anymore

rhaschke, 2021-12-16:
In the past it was possible to switch between the different tutorial variants of the ROS distro. Now the link from https://moveit.ros.org points here: https://moveit.picknik.ai and there is no dropdown to switch. Please make the ROS1 tutorials accessible again.

gavanderhoorn, 2021-12-16:
It's a bit hidden, but the "drop down" (which is more a "drop up" now) is in the bottom-left corner.

Presumably this was done to make MoveIt 2's documentation look more like the ROS 2 documentation site.

The previous layout with the clearly marked-as-such version selector in the top-left was probably easier for users to find though ..

rhaschke, 2021-12-16:
Thanks for the hint! I really looked hard everywhere. I suggest placing the selector more prominently again.

tylerjw, 2021-12-16:
The move away from the drop-down came with an upgrade in the version of sphinx that we are using to generate the ros2 versions.  This upgrade made so we were able to use the multiversion plugin that lets us maintain multiple versions of the tutorials (for the different ros2 versions) in separate branches of the same repo.  With the ros1 version, each version of that site is hosted separately and is generated from separate repos.

The change to the selector being at the bottom right makes the docs website less custom and more standardized.  It is in-line with the https://docs.ros.org/en/rolling/ site and many similar sites.

gavanderhoorn, 2021-12-16:
Which does not necessarily mean it results in a better UX @tylerjw.

Consistency is certainly worthwhile, but I don't believe anyone can claim that this:

[image]

is as visible and clear as:

[image]

tylerjw, 2021-12-16:
You are welcome to help me try to add it back.  

No one seemed to care about the moveit2 tutorial website and the user experience of following the tutorials themselves.  

I've been trying to make so the content itself is truthful and gives a user as clear of a guide for using what does work in moveit2 as I can.  I honestly care much less about the user experience for switching ROS versions than I do about the user experience of trying to read and understand how to use moveit based on the tutorials.

This UI element was a casualty of me upgrading the version so I could use multiversion.  I didn't have the political capitol to delete tutorials out of the main branch that are broken so I created a foxy version and hardened that because I assume that is where most of the new ros2 users will start.

tylerjw, 2021-12-16:
The priorities of those working on the tutorials here over the last couple of years have been all out of whack.  I'm sorry if I responded harshly to this.  I think we should care much more about how difficult it is to read the tutorials and use MoveIt than things like how the site looks.  I think the primary reason we nitpick things like this to death and don't focus on things like the voice and content of the tutorials staying up to date is it is much easier to have a strong opinion about something simple like this.

I don't have time now but if no one else fixes this before Humble I'll add it to the list of things I'll try to get done for the tutorial website before Humble.

gavanderhoorn, 2021-12-17:
There was no criticism directed to you specifically @tylerjw.

I fully appreciate the difficulties of finding sufficient time to work on things.

>  I think we should care much more about how difficult it is to read the tutorials and use MoveIt than things like how the site looks.

Fully agreed.

> You are welcome to help me try to add it back.

doing that here will cause divergence between `ros2_documentation` and the MoveIt 2 tutorial site, as it will probably require a template change.

rhaschke, 2021-12-17:
Dear Tyler, I fully second Gijs comment. There was no particular criticism to anybody. I just didn't find the switching drop-down at all. Having it top-right again, would probably help users to find it. If that's too complicated now, because you switched templates, it is probably not worth the effort. However, if there is an easy solution, the website experts should go for it.

tylerjw, 2021-12-17:
I'm sorry for responding the way I did.  I'll try to figure out how to add it back.  I agree it is more ergonomic.

tylerjw, 2023-12-01:
The current state of the site has a link for the moveit1 tutorials. I never got the drop-down back, and the versioning of the site went away too.

[image]

closed 2023-12-01 by tylerjw