moveit/moveit2_tutorials #285: Remove dead key-value pairs from `extlinks` in `conf.py`

mlautman, 2022-01-14:
### Description

There are a number of dead links in `extlinks` in the `conf.py` file that should be removed (e.g., anything that points to `"http://docs.ros.org/" + ros1_distro + "/api/` is dead for ROS 2). A meaningful contribution would be to first remove all of the dead key-value pairs in a PR, and in follow on PR, start the process of unifying links distributed across the tutorials with new key-value pairs (e.g., [kinematic_constraints](https://github.com/ros-planning/moveit2_tutorials/pull/250#discussion_r785132288))

mlautman, 2022-01-14:
Tagging @corycrean and @tylerjw for visibility

vatanaksoytezer, 2022-01-14:
I am currently working on having a moveit.picknik.ai/api that will contain API documentation for MoveIt 2. Hopefully we will just change `http://docs.ros.org/" + ros1_distro +` with `moveit.picknik.ai/api` should suffice.

closed 2023-12-01 by tylerjw