moveit/moveit2_tutorials #308: Move Group Interface tutorial not running on Galactic

tlpss, 2022-01-17:
### Description

I believe the move Group interface tutorial does not run on the latest galactic release of Moveit2. When starting the tutorial launch file I get following warning, after which the code seems to block during the MGI constructor call:

```
root@tlips-XPS:/workspaces/galactic_colcon_ws/ROS# ros2 launch moveit2_tutorials move_group_interface_tutorial.launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2022-01-17-19-41-12-727878-tlips-XPS-29265
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group_interface_tutorial-1]: process started with pid [29270]
[move_group_interface_tutorial-1] [WARN] [1642448472.961407490] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
```

2 sidenotes:
- I manually removed moveit from the .repos file to avoid moveit being built from source 
- I temporarily disabled the moveit_servo as I got a depraction error.

### Your environment
* ROS Distro: Galactic
* OS Version: Ubuntu 20.04
* Source or Binary build? Binary using the latest galactic-release docker image from moveit2.

### Steps to reproduce
- pull the docker container
- follow the instructions for cloning this repo, installing dependencies and building the packages
- launch the move group and move group tutorial as explained in the tutorial instructions.

### Expected behaviour
The MGI tutorial should start and display trajectories as in Foxy. I followed the same procedure for foxy and the tutorial worked as expected.

vatanaksoytezer, 2022-01-17:
Hello @tlpss,

Thanks for reporting the issue. I've a few questions to understand the problem more in depth:

- Which docker container are you pulling?
- Which instructions are you following?
- Is that all the logging you get or are there more logging of the terminal output? If not can you share the full log?

A few things:

- Can you try installing moveit with `sudo apt-get install ros-galactic-moveit`. The binary build is broken in Galactic right now (we are working on a patch), so if this command gives you errors, you might have to source built for now.
- @tylerjw I think we should remove moveit2 from .repos anyway.

tlpss, 2022-01-17:
Thanks for the quick reply @vatanaksoytezer and apologies for not being precise enough.

> Which docker container are you pulling?

[latest galactic release container on the moveit2 docker hub](https://hub.docker.com/layers/moveit/moveit2/galactic-ci-testing/images/sha256-d8e707f8a9043b157d4de5a1bfcdca7e3d5ca33aacf97a1f061d26fe873506bd?context=explore)

> Which instructions are you following?

https://moveit.picknik.ai/foxy/doc/getting_started/getting_started.html# where I changed the branch of the moveit2 tutorials from foxy to galactic.

> Is that all the logging you get or are there more logging of the terminal output? If not can you share the full log?

That is unfortunately al the logging I get, after that the executor is not responsive ( even trying to kill it with CTRL + C does not work). 

> Can you try installing moveit with sudo apt-get install ros-galactic-moveit

This gives  
```
root@tlips-XPS:/workspaces/galactic_colcon_ws/ROS# sudo apt-get install ros-galactic-moveit
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ros-galactic-moveit is already the newest version (2.3.0-1focal.20211223.045810).
0 upgraded, 0 newly installed, 0 to remove and 11 not upgraded.
```

vatanaksoytezer, 2022-01-18:
@tlpss I can't do `sudo apt-get install ros-galactic-moveit `in the latest version of docker right now because the builds are broken (patch is already merged we will release that as soon as possible). Assuming you changed the branch name to galactic in https://moveit.picknik.ai/foxy/doc/getting_started/getting_started.html#download-moveit-2-tutorials-source and sourced galactic instead of foxy in https://moveit.picknik.ai/foxy/doc/getting_started/getting_started.html#download-moveit-2-tutorials-source just following the instructions should suffice as long as you are using the source version of moveit2 (ie: don't remove moveit2 from the repos file).

I just tested the build instructions (interchanging foxy with galactic) with the latest docker image tagged as `moveit/moveit2/galactic-ci-testing` and seemed to work just fine. Let me know if you can reproduce the tutorials with a source build of moveit2, I don't expect a released version of moveit for Galactic or Rolling would do the job right now.

Please don't hesitate to comment here if you have more questions., and we will be happy to help!

tlpss, 2022-01-18:
@vatanaksoytezer , thanks for you reply. I can confirm that I can't do the apt install. Will try to build moveit from source in the galactic-source container and check if I can reproduce the tutorials. 

>  because the builds are broken 

Is there any way to get notified for such events? (mailing list or something)

vatanaksoytezer, 2022-01-18:
> Is there any way to get notified for such events? (mailing list or something)

Just to clarify this is not something usually happens, but you can always track our buildfarm status in the very bottom of moveit2 README: https://github.com/ros-planning/moveit2.

tlpss, 2022-01-18:
@vatanaksoytezer, the tutorial is working as expected after I built moveit from source. 

Thanks for your help!

closed 2022-01-18 by tlpss

tlpss, 2022-01-25:
@vatanaksoytezer , I noticed that the patch was released and that the latest release for galactic is now 2.3.3-1.

However after running `apt-get update` in the docker container, the Moveit version that apt tries to install is 2.3.0-1. Any suggestions on how to install the latest release?

vatanaksoytezer, 2022-01-25:
Hey @tlpss unfortunately the buildfarm is still broken due to a dependency issue, we might release another patch very soon. Until then I would advice to stick with the source build.