moveit/moveit2_tutorials #329: MoveIt Visual Tools does not develop for ros foxy

AVJF1996, 2022-02-12:
Hi everybody 
MoveIt Visual Tools doesn't develop for ros foxy so please omit that from dependencies or think for a way because I have been struggling for 5 hours 
the last solution which I tried 
making  install MoveIt Visual Tools and finally, colcon build error to me that I haven't known rocpp 
if I source ros1 directory everything goes mess and I got lots of errors

tylerjw, 2022-02-13:
It does build on foxy.  You just have to make sure you are checking out the right branch.  There is a foxy branch of this repo with this repos file:

https://github.com/ros-planning/moveit2_tutorials/blob/foxy/moveit2_tutorials.repos

AVJF1996, 2022-02-23:
> It does build on foxy. You just have to make sure you are checking out the right branch. There is a foxy branch of this repo with this repos file:
> 
> https://github.com/ros-planning/moveit2_tutorials/blob/foxy/moveit2_tutorials.repos

Thanks a lot friend but you know still there is some problem I think it is better to add these packages to moveit2_tutrials.repos or something like dependencies because I installed dependencies but it didn't install these two packages for me and one package which called msg-graphs or something like this I downloaded that with my self and add it to src to work for me 
Thanks a lot for your clear answer

closed 2022-02-23 by AVJF1996