moveit/moveit2_tutorials #337: MoveIt Commander with ROS2

Maltergate, 2022-02-25:
### Description

Hi,
Thank you very much for the wonderful work. I'm trying to use ROS2 as part of my research using a UR5 Robotic arm. More specifically, I'd like to set some pose (position, orientation) of the wrist of the robot. I've been able to play a bit with Moveit2 in Rviz, it works like a charm.

I now would like to specify, plan and go to the specific pose in a python script (so I can loop over different poses). Following tutorial, I tried to use the python binding interface, but is uses rospy, which is ROS1, not ROS2. I also tried to use the moveit_commander but couldn't find how to launch it. Is the moveit_commander available for ROS2 ?

Thanks.

### Your environment
* ROS2 Galactic
* Ubuntu 20.04

### Steps to reproduce
Following the tutorial for python bindings

v4hn, 2022-02-28:
> I also tried to use the moveit_commander but couldn't find how to launch it. Is the moveit_commander available for ROS2 ?

No, not at this point. You can probably use the move_group's action API for what you want to do, but you will need to call the action yourself as no python wrappers are available yet.

ozangungortuhh, 2022-03-15:
Hello @v4hn, can you please give a little bit more detail regarding your answer. I am trying to migrate from ROS1 to ROS2 and my main challenge is the same as @Maltergate, that I am not able to find any tutorial or any kind of material to be able to move the robot with Python ROS2 nodes.

samiulextreem, 2022-05-26:
Moveit commander is not available yet for ros2. I know it a bummer. The doc is also kinda messy amd does notention that properly.

peterdavidfagan, 2023-10-12:
Closing as `moveit_py` is now available. Feel free to post feature requests for `moveit_py` features that are not currently available.

closed 2023-10-12 by peterdavidfagan

Darshi18, 2024-02-07:
how to use pymoveit2 and how install that 
when i try to intsall 
pip install pymoveit2
Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not find a version that satisfies the requirement pymoveit2 (from versions: none)
ERROR: No matching distribution found for pymoveit2

emp002, 2024-04-18:
> @Maltergate This repo allows you to interact with the MoveIt 2 commands in Python and ROS 2: https://github.com/AndrejOrsula/pymoveit2
> 
> Official python support might be available after the Google Summer of Code 🚀 https://moveit.ros.org/events/moveit/mentor/google/2022/05/20/2022-google-summer-of-code-students.html

I am trying to use the above package for my 4Dof Scara arm. Below I have attached the mode l of my robot.
First, I tried using ex_fk.py to get end effector position by giving joint angles as arguments.
Then, I used the same values with 12 decimal places to get joint angles using ex_ik.py. It works perfectly, it gives joint angles close to [0, 0, 0, 0] but with decimal places.
Then again, I used the same values but rounded-off to 1 decimal. I am getting an error that "IK computation failed! Error code: -31."

I have also attached the screenshot of my robot description terminal and the ex_ik launch terminal.
What could be the reason that it is not calculating IK with less decimal places?

[image]
[image]
[image]

aymanT122R, 2025-04-15:
> Moveit commander is not available yet for ros2. I know it a bummer. The doc is also kinda messy amd does notention that properly.

any update for moveit_commander with ros2