moveit/moveit2_tutorials #338: ERROR: the following packages/stacks could not have their rosdep keys resolved to system dependencies

DolceParadise, 2022-02-28:
### Description

While attempting to install MoveIt2 on ROS Rolling from the instructions at this [tutorial](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html), I encountered the following errors. 
Running `rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y` returns the following output. Do note that all the steps before running this were successful and ROS2 Rolling was sourced. 
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
moveit_common: No definition of [backward_ros] for OS version [focal]
moveit2_tutorials: No definition of [warehouse_ros_mongo] for OS version [focal]
moveit_hybrid_planning: No definition of [ros_testing] for OS version [focal]
pilz_industrial_motion_planner: No definition of [orocos_kdl] for OS version [focal]
moveit_core: No definition of [ruckig] for OS version [focal]
moveit_ros_planning: No definition of [ros_testing] for OS version [focal]
moveit_ros_planning_interface: No definition of [ros_testing] for OS version [focal]
moveit_kinematics: No definition of [orocos_kdl] for OS version [focal]
moveit_servo: No definition of [ros_testing] for OS version [focal]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```
Following this, running `colcon build --mixin release` returns the following while building moveit_msgs 
```
fatal error: object_recognition_msgs/msg/detail/object_type__struct.h: No such file or directory

   57 | #include "object_recognition_msgs/msg/detail/object_type__struct.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/moveit_msgs__rosidl_generator_c.dir/build.make:1452: CMakeFiles/moveit_msgs__rosidl_generator_c.dir/rosidl_generator_c/moveit_msgs/msg/detail/attached_collision_object__functions.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
```
The final output of the colcon build command is this 
```
Failed   <<< moveit_msgs [30.7s, exited with code 2]
Aborted  <<< rviz_visual_tools [37.0s]             
Aborted  <<< geometric_shapes [1min 7s]                                     

Summary: 10 packages finished [1min 8s]
  1 package failed: moveit_msgs
  2 packages aborted: geometric_shapes rviz_visual_tools
  3 packages had stderr output: geometric_shapes moveit_msgs rviz_visual_tools
  32 packages not processed
```

### Your environment
* ROS Distro: Rolling
* OS Version: Ubuntu 20.04
* Source build

### Steps to reproduce
Follow along with the tutorials at [MoveIt Rolling Ridley Getting Started](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html). 

### Expected behaviour
moveit_tutorials and its contemporary packages should have been built successfully.

vatanaksoytezer, 2022-03-02:
Thank you for reporting @DolceParadise.

You can still use rolling and rosdep on focal with: `export ROSDISTRO_INDEX_URL=https://raw.githubusercontent.com/ros/rosdistro/rolling/2022-01-28/index-v4.yaml` that changes the rosdistro database.

Having said that galactic should work just fine. Make sure you are only sourcing galactic, clean your workspace (by removing build install and log folders) and recompile again. Please let me know if the issue persists.

tylerjw, 2022-03-05:
I think I've seen that error when my machine runs out of ram building.  Have you tried building with less threads or watched htop or something similar to see if that's what is happening?

vatanaksoytezer, 2022-05-17:
This issue seems stalled, and should be resolved with @tylerjw's suggestion. Please feel free to re-open if you are having similar issues.

closed 2022-05-17 by vatanaksoytezer