moveit/moveit2_tutorials #342: Error launching moveit2_tutorials

srirsank, 2022-03-24:
### Description

I just installed ROS2 and MoveIt2 by following instructions listed at https://moveit.picknik.ai/galactic/doc/tutorials/tutorials.html. But after installing, I encounter errors while running the very first tutorial.

I have tried installing ROS2 using binaries, build from source, as well as in a docker container; same end result in all cases.

Digging into this, I see that the Panda trajectory controller's filename was updated for ROS2, but demo.launch.py doesn't reflect this. Patch for the proposed fix is pasted below.

Proposed Fix:
diff --git a/doc/tutorials/quickstart_in_rviz/launch/demo.launch.py b/doc/tutorials/quickstart_in_rviz/launch/demo.launch.py
index 7800d5da7..a097f0491 100644
--- a/doc/tutorials/quickstart_in_rviz/launch/demo.launch.py
+++ b/doc/tutorials/quickstart_in_rviz/launch/demo.launch.py
@@ -76,7 +76,7 @@ def generate_launch_description():
 
     # Trajectory Execution Functionality
     moveit_simple_controllers_yaml = load_yaml(
-        "moveit_resources_panda_moveit_config", "config/panda_controllers.yaml"
+        "moveit_resources_panda_moveit_config", "config/panda_ros2_controllers.yaml"
     )
     moveit_controllers = {
         "moveit_simple_controller_manager": moveit_simple_controllers_yaml,

### Your environment
* ROS Distro: Galactic
* OS Version: Ubuntu 20.04
* Source or Binary build? Tried all with same results
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
Installed ROS2 from binaries by following instructions at https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html
Installed ROS2 from source by following instructions at https://docs.ros.org/en/galactic/Installation/Ubuntu-Development-Setup.html
Used a docker container for ROS2 as described at https://docs.ros.org/en/galactic/How-To-Guides/Run-2-nodes-in-single-or-separate-docker-containers.html

In all three cases, after setting up ROS2/colcon and sourcing the colcon workspace, I see the following when I launch the demo:
# ros2 launch moveit2_tutorials demo.launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2022-03-24-22-56-25-962037-d940494417b7-76142
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py:226> exception=InvalidLaunchFileError('py')>
Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 53, in get_launch_description_from_any_launch_file
    return loader(launch_file_path)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 68, in get_launch_description_from_python_launch_file
    return getattr(launch_file_module, 'generate_launch_description')()
  File "/root/ws_moveit2/install/moveit2_tutorials/share/moveit2_tutorials/launch/demo.launch.py", line 101, in generate_launch_description
    run_move_group_node = Node(
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/actions/node.py", line 206, in __init__
    normalized_params = normalize_parameters(parameters)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/utilities/normalize_parameters.py", line 184, in normalize_parameters
    normalized_params.append(normalize_parameter_dict(param))
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/utilities/normalize_parameters.py", line 165, in normalize_parameter_dict
    raise TypeError('Unexpected type for parameter value {}'.format(repr(value)))
TypeError: Unexpected type for parameter value None

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 127, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_description_source.py", line 53, in _get_launch_description
    return get_launch_description_from_any_launch_file(location)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 56, in get_launch_description_from_any_launch_file
    raise InvalidLaunchFileError(extension, likely_errors=exceptions)
launch.invalid_launch_file_error.InvalidLaunchFileError: Caught exception when trying to load file of format [py]: Unexpected type for parameter value None

### Expected behaviour
RViz should pop up

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

RoboHack04, 2022-03-26:
@srirsank I am glad to find this issue. Even I was facing the same issue while working on the tutorials today. However, I believe there are two file name changes to be made like suggested below -

    # diff --git a/doc/tutorials/quickstart_in_rviz/launch/demo.launch.py b/doc/tutorials/quickstart_in_rviz/launch/demo.laun
    ch.py
    index 7800d5da7..64326c720 100644
    --- a/doc/tutorials/quickstart_in_rviz/launch/demo.launch.py
    +++ b/doc/tutorials/quickstart_in_rviz/launch/demo.launch.py
    @@ -76,7 +76,7 @@ def generate_launch_description():
     
         # Trajectory Execution Functionality
         moveit_simple_controllers_yaml = load_yaml(
    -        "moveit_resources_panda_moveit_config", "config/panda_controllers.yaml"
    +        "moveit_resources_panda_moveit_config", "config/panda_moveit_controllers.yaml"
         )
         moveit_controllers = {
             "moveit_simple_controller_manager": moveit_simple_controllers_yaml,
    @@ -169,7 +169,7 @@ def generate_launch_description():
         ros2_controllers_path = os.path.join(
             get_package_share_directory("moveit_resources_panda_moveit_config"),
             "config",
    -        "panda_ros_controllers.yaml",
    +        "panda_ros2_controllers.yaml",
         )--

# Problem while moving robot
Even after making these changes the spawned panda robot is starting in a collusion state and hence it doesn't allow to plan any goal commands. Let me know if you are also facing this problem.

NeoKish, 2022-04-07:
Hi @srirsank @RoboHack04 , I tried building from main branch 

`git clone https://github.com/ros-planning/moveit2_tutorials.git --branch main`

ROS distro: Galactic
Ubuntu: 20.04

This seems to be the updated one (recent update has been 2 days ago) and it works. 

[image]

one thing you might have to do during colcon build is to allow overrididing the other installs so you have successful build without any '--stderr' errors

jomoengineer, 2022-05-04:
I attempted the same as the OP and ran into the same issue.  I was following the Galactic MoveIt2 install and cloned the Galactic Tutorial branch and hit the same InvalidLaunchFileError error.

vatanaksoytezer, 2022-05-17:
@tylerjw  made a recent walkthrough on tutorials and seemed fine with the latest instructions. Can you guys @srirsank @NeoKish @RoboHack04 @jomoengineer can you still confirm if you are having similar issues?

RoboHack04, 2022-05-17:
@vatanaksoytezer yes, the latest instructions seems to work for me, however for anyone facing problems...i would say building the package with 'colcon build --mixin release' and ensuring no stderrors after build is very important step.

vatanaksoytezer, 2022-05-25:
Closing this issue since it seems latest instructions are working for people. Feel free to reopen this issue.

closed 2022-05-25 by vatanaksoytezer