moveit/moveit2_tutorials #350: has no member named ‘future’ when building moveit2_tutorials

valdesreciof, 2022-04-26:
### Description

When following the moveit2 Getting Started tutorial, executing the "colcon build --mixin release" command results in a "has no member named ‘future’" error for the moveit2_tutorials package. This has happened while trying to set up moveit on two separate machines now. I have tried to add "#include <future>" to /ws_moveit2/src/moveit2_tutorials/doc/examples/planning_scene_ros_api/src/planning_scene_ros_api_tutorial.cpp and the console spits out the same error.

### Your environment
* ROS Distro: Galactic
* OS Version: Ubuntu 20.04
* Binary build

### Steps to reproduce
install galactic following these steps: https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html
then follow moveit setup guide here: https://moveit.picknik.ai/galactic/doc/tutorials/getting_started/getting_started.html

### Expected behaviour
all packages should build without error 

### Backtrace or Console output

```
desktop:~/ws_moveit2$ colcon build --mixin release
[0.433s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'moveit_resources_panda_description' is in: /opt/ros/galactic
	'moveit_resources_panda_moveit_config' is in: /opt/ros/galactic
	'launch_param_builder' is in: /opt/ros/galactic
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding launch_param_builder moveit_resources_panda_description moveit_resources_panda_moveit_config

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> moveit_resources_panda_description
Starting >>> moveit_resources_fanuc_description
Starting >>> moveit_resources_pr2_description
Starting >>> moveit_visual_tools
Starting >>> launch_param_builder
Finished <<< moveit_resources_panda_description [0.94s]                                                                                                                                                                                                       
Finished <<< moveit_resources_fanuc_description [0.93s]
Starting >>> moveit_resources_panda_moveit_config
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_visual_tools [0.94s]
Finished <<< moveit_resources_pr2_description [0.96s]
Finished <<< moveit_resources_panda_moveit_config [0.17s]                                                                                                                  
Starting >>> moveit2_tutorials
Finished <<< moveit_resources_fanuc_moveit_config [0.18s]
Starting >>> moveit_resources
Finished <<< moveit_resources [0.13s]                                                                                           
--- stderr: launch_param_builder                                                                    
/home/fernando/.local/lib/python3.8/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< launch_param_builder [3.20s]
--- stderr: moveit2_tutorials                             
/home/fernando/ws_moveit2/src/moveit2_tutorials/doc/examples/planning_scene_ros_api/src/planning_scene_ros_api_tutorial.cpp: In function ‘int main(int, char**)’:
/home/fernando/ws_moveit2/src/moveit2_tutorials/doc/examples/planning_scene_ros_api/src/planning_scene_ros_api_tutorial.cpp:164:77: error: ‘using SharedFuture = class std::shared_future<std::shared_ptr<moveit_msgs::srv::ApplyPlanningScene_Response_<std::allocator<void> > > >’ {aka ‘class std::shared_future<std::shared_ptr<moveit_msgs::srv::ApplyPlanningScene_Response_<std::allocator<void> > > >’} has no member named ‘future’
  164 |   response_future = planning_scene_diff_client->async_send_request(request).future.share();
      |                                                                             ^~~~~~
make[2]: *** [doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/build.make:63: doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/src/planning_scene_ros_api_tutorial.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:380: doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< moveit2_tutorials [6.32s, exited with code 2]

Summary: 8 packages finished [7.68s]
  1 package failed: moveit2_tutorials
  2 packages had stderr output: launch_param_builder moveit2_tutorials
```

elikh4, 2022-04-27:
https://github.com/ros-planning/moveit2_tutorials/pull/198
the issue was solved in this post see the files changed for the fix

vatanaksoytezer, 2022-05-17:
I think this can be resolved with #198. Please feel free to re-open if you are still seeing this.

closed 2022-05-17 by vatanaksoytezer