moveit/moveit2_tutorials #364: Planning For Differential-Drive Mobile Base and an Arm tutorial is broken

TomGrimwood, 2022-05-14:
### Description

This tutorial has a lot of broken repository references and I am unable to get it working.

### Your environment
* ROS Distro: Galactic
* OS Version: Ubuntu 20.04
* Source

### Steps to reproduce
```Shell
cd ~/ws_moveit/src
wget https://raw.githubusercontent.com/PickNikRobotics/stretch_ros/ros2/stretch_ros.repos
vcs import < stretch_ros.repos
```
stretch_ros.repos references version foxy which does not exist, should it be ros2-beta?
```
  realsense-ros:  # For realsense2_description
    type: git
    url: https://github.com/IntelRealSense/realsense-ros
    version: foxy
```

```Shell
rosdep install -r --from-paths . --ignore-src --rosdistro galactic -y
```
results in

```Shell
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
stretch_moveit_config: Cannot locate rosdep definition for [moveit_fake_controller_manager]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

I am also unable to build these two packages from DLu's branch, should they be changed or is it required to use this branch?
```Shell
  moveit2:
    type: git
    url: https://github.com/DLu/moveit2
    version: dlu_galactic

  moveit_task_constructor:
    type: git
    url: https://github.com/DLu/moveit_task_constructor
    version: galactic
```

```Shell
--- stderr: moveit_core
/usr/bin/ld: libmoveit_distance_field.so.2.1.3: undefined reference to `boost::iostreams::zlib::finish'
/usr/bin/ld: libmoveit_distance_field.so.2.1.3: undefined reference to `boost::iostreams::zlib::no_flush'
/usr/bin/ld: libmoveit_distance_field.so.2.1.3: undefined reference to `boost::iostreams::detail::zlib_base::xdeflate(int)'
/usr/bin/ld: libmoveit_distance_field.so.2.1.3: undefined reference to `boost::iostreams::detail::zlib_base::before(char const*&, char const*, char*&, char*)'
/usr/bin/ld: libmoveit_distance_field.so.2.1.3: undefined reference to `boost::iostreams::zlib_error::check(int)'
/usr/bin/ld: libmoveit_distance_field.so.2.1.3: undefined reference to `boost::iostreams::zlib::default_compression'
/usr/bin/ld: libmoveit_distance_field.so.2.1.3: undefined reference to `boost::iostreams::detail::zlib_base::do_init(boost::iostreams::zlib_params const&, bool, void* (*)(void*, unsigned int, unsigned int), void (*)(void*, void*), void*)'
/usr/bin/ld: libmoveit_distance_field.so.2.1.3: undefined reference to `boost::iostreams::zlib::stream_end'
/usr/bin/ld: libmoveit_distance_field.so.2.1.3: undefined reference to `boost::iostreams::detail::zlib_base::zlib_base()'
/usr/bin/ld: libmoveit_distance_field.so.2.1.3: undefined reference to `boost::iostreams::detail::zlib_base::~zlib_base()'
/usr/bin/ld: libmoveit_distance_field.so.2.1.3: undefined reference to `boost::iostreams::detail::zlib_base::after(char const*&, char*&, bool)'
/usr/bin/ld: libmoveit_distance_field.so.2.1.3: undefined reference to `boost::iostreams::zlib::default_strategy'
/usr/bin/ld: libmoveit_distance_field.so.2.1.3: undefined reference to `boost::iostreams::detail::zlib_base::reset(bool, bool)'
/usr/bin/ld: libmoveit_distance_field.so.2.1.3: undefined reference to `boost::iostreams::zlib::sync_flush'
/usr/bin/ld: libmoveit_distance_field.so.2.1.3: undefined reference to `boost::iostreams::detail::zlib_base::xinflate(int)'
/usr/bin/ld: libmoveit_distance_field.so.2.1.3: undefined reference to `boost::iostreams::zlib::deflated'
collect2: error: ld returned 1 exit status
make[2]: *** [distance_field/CMakeFiles/test_distance_field.dir/build.make:218: distance_field/test_distance_field] Error 1
make[1]: *** [CMakeFiles/Makefile2:2207: distance_field/CMakeFiles/test_distance_field.dir/all] Error 2
[... 42 lines cut ...]
[move_group-1] [ERROR] [1652569099.293457009] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/geometric/planners/rrt/src/RRTstar.cpp:193 - stretch_arm/stretch_arm[RRTstar]: There are no valid initial states!
[move_group-1] [WARN] [1652569099.293486985] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/tools/multiplan/src/ParallelPlan.cpp:138 - ParallelPlan::solve(): Unable to find solution by any of the threads in 0.000679 seconds
[move_group-1] [WARN] [1652569099.299217969] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/base/src/Planner.cpp:257 - stretch_arm/stretch_arm[RRTstar]: Skipping invalid start state (invalid state)
[move_group-1] [ERROR] [1652569099.299258417] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/geometric/planners/rrt/src/RRTstar.cpp:193 - stretch_arm/stretch_arm[RRTstar]: There are no valid initial states!
[move_group-1] [WARN] [1652569099.299301364] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/base/src/Planner.cpp:257 - stretch_arm/stretch_arm[RRTstar]: Skipping invalid start state (invalid state)
[move_group-1] [ERROR] [1652569099.299335689] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/geometric/planners/rrt/src/RRTstar.cpp:193 - stretch_arm/stretch_arm[RRTstar]: There are no valid initial states!
[move_group-1] [WARN] [1652569099.299943659] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/base/src/Planner.cpp:257 - stretch_arm/stretch_arm[RRTstar]: Skipping invalid start state (invalid state)
[move_group-1] [ERROR] [1652569099.299994654] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/geometric/planners/rrt/src/RRTstar.cpp:193 - stretch_arm/stretch_arm[RRTstar]: There are no valid initial states!
[move_group-1] [WARN] [1652569099.300013331] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/base/src/Planner.cpp:257 - stretch_arm/stretch_arm[RRTstar]: Skipping invalid start state (invalid state)
[move_group-1] [ERROR] [1652569099.300047104] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/geometric/planners/rrt/src/RRTstar.cpp:193 - stretch_arm/stretch_arm[RRTstar]: There are no valid initial states!
[move_group-1] [WARN] [1652569099.300094019] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/tools/multiplan/src/ParallelPlan.cpp:138 - ParallelPlan::solve(): Unable to find solution by any of the threads in 0.000944 seconds
[move_group-1] [WARN] [1652569099.300221809] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/base/src/Planner.cpp:257 - stretch_arm/stretch_arm[RRTstar]: Skipping invalid start state (invalid state)
[move_group-1] [ERROR] [1652569099.300250027] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/geometric/planners/rrt/src/RRTstar.cpp:193 - stretch_arm/stretch_arm[RRTstar]: There are no valid initial states!
[move_group-1] [WARN] [1652569099.300290189] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/base/src/Planner.cpp:257 - stretch_arm/stretch_arm[RRTstar]: Skipping invalid start state (invalid state)
[move_group-1] [ERROR] [1652569099.300313882] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/geometric/planners/rrt/src/RRTstar.cpp:193 - stretch_arm/stretch_arm[RRTstar]: There are no valid initial states!
[move_group-1] [WARN] [1652569099.300345987] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/tools/multiplan/src/ParallelPlan.cpp:138 - ParallelPlan::solve(): Unable to find solution by any of the threads in 0.000171 seconds
[move_group-1] [INFO] [1652569099.300399444] [moveit.ompl_planning.model_based_planning_context]: Unable to solve the planning problem
[move_group-1] [INFO] [1652569099.300455636] [moveit_move_group_default_capabilities.move_action_capability]: No motion plan found. No execution attempted.
[rviz2-2] [INFO] [1652569099.594736787] [move_group_interface]: Planning request aborted
[rviz2-2] [ERROR] [1652569099.694921828] [move_group_interface]: MoveGroupInterface::plan() failed or timeout reached
```
Is this due to swapping out to many repositories changing the expected behaviour? I am certainly out of my depth.

Thanks,
Tom.

garyraise, 2022-08-10:
Has there been any update on this issue? I'm having the same problem

tylerjw, 2023-12-01:
This is a very experimental feature. Improvements to this would be really appreciated.

tonitonitonitoni, 2025-03-05:
Still broken in March 2025! Please fix this it is so important