moveit/moveit2_tutorials #388: MoveIt 2 Humble Getting Started colcon build results in CMake Error at CMakeLists.txt:6

jomoengineer, 2022-05-25:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source or Binary build? Source
* If binary, which release version?
* If source, which git commit or tag? latest

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

While running through the MoveIt 2 Humble Getting Started Tutorial, the colcon build step fails running the following:

```
$ colcon build --mixin release
```
### Expected behaviour
It should build without error

### Backtrace or Console output
```
~/ws_moveit2$ colcon build --mixin release
Starting >>> srdfdom 
Starting >>> moveit_common
Starting >>> moveit_resources_panda_description
Starting >>> launch_param_builder
Finished <<< moveit_common [0.61s]                                                          
Starting >>> moveit_resources_fanuc_description
--- stderr: launch_param_builder                             
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< launch_param_builder [0.97s]
Starting >>> moveit_task_constructor_msgs                    
Finished <<< moveit_resources_panda_description [1.01s]
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_fanuc_description [0.83s]      
Starting >>> rviz_marker_tools
--- stderr: moveit_task_constructor_msgs                     
CMake Error at CMakeLists.txt:6 (find_package):
  By not providing "Findmoveit_msgs.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "moveit_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "moveit_msgs" with
  any of the following names:

    moveit_msgsConfig.cmake
    moveit_msgs-config.cmake

  Add the installation prefix of "moveit_msgs" to CMAKE_PREFIX_PATH or set
  "moveit_msgs_DIR" to a directory containing one of the above files.  If
  "moveit_msgs" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< moveit_task_constructor_msgs [0.70s, exited with code 1]
Aborted  <<< moveit_resources_panda_moveit_config [0.68s]               
Aborted  <<< srdfdom [1.78s]
Aborted  <<< rviz_marker_tools [1.66s]                            

Summary: 4 packages finished [3.42s]
  1 package failed: moveit_task_constructor_msgs
  3 packages aborted: moveit_resources_panda_moveit_config rviz_marker_tools srdfdom
  2 packages had stderr output: launch_param_builder moveit_task_constructor_msgs
  42 packages not processed

```
Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

vatanaksoytezer, 2022-05-25:
Hello, @jomoengineer thank you for reporting an issue. Any chance you missed the rosdep installation step as in: `rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y`?

Dou you mind sharing the output of rosdep?

jomoengineer, 2022-05-26:
This is my command history from running through the Getting Started guide:

```
. ~/ros2_humble/ros2-linux/setup.bash
sudo apt-get update
sudo apt install python3-rosdep
sudo rosdep init
rosdep update
sudo apt update
sudo apt dist-upgrade
sudo apt install python3-colcon-common-extensions
sudo apt install python3-colcon-mixin
colcon mixin add default https://raw.githubusercontent.com/colcon/colcon-mixin-repository/master/index.yaml
colcon mixin update default
sudo apt install python3-vcstool
mkdir -p ~/ws_moveit2/src
cd ws_moveit2/
cd src/
git clone https://github.com/ros-planning/moveit2_tutorials -b main
vcs import < moveit2_tutorials/moveit2_tutorials.repos
rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
cd ..
colcon build --mixin release
```

This is the rosdep version I am running. Just running rosdep just gave an error since it is expecting a parameter.
```
$ rosdep --version
0.21.0
```

jomoengineer, 2022-05-26:
Well, here is the output of the rosdep command you suggested I missed.  It appears there is an issue here though:
```
$ rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
moveit_hybrid_planning: Cannot locate rosdep definition for [std_srvs]
moveit_ros_benchmarks: Cannot locate rosdep definition for [ament_lint_common]
moveit: Cannot locate rosdep definition for [ament_lint_common]
moveit_resources_prbt_ikfast_manipulator_plugin: Cannot locate rosdep definition for [tf2_kdl]
moveit_ros_perception: Cannot locate rosdep definition for [ament_lint_common]
moveit_configs_utils: Cannot locate rosdep definition for [ament_lint_common]
pilz_industrial_motion_planner_testutils: Cannot locate rosdep definition for [ament_lint_common]
moveit_plugins: Cannot locate rosdep definition for [ament_lint_common]
moveit_visual_tools: Cannot locate rosdep definition for [ament_lint_common]
moveit_simple_controller_manager: Cannot locate rosdep definition for [ament_lint_common]
moveit_ros: Cannot locate rosdep definition for [ament_lint_common]
rosparam_shortcuts: Cannot locate rosdep definition for [ros_testing]
moveit_resources_prbt_moveit_config: Cannot locate rosdep definition for [robot_state_publisher]
moveit_ros_control_interface: Cannot locate rosdep definition for [ament_lint_common]
moveit_kinematics: Cannot locate rosdep definition for [tf2_kdl]
moveit_core: Cannot locate rosdep definition for [ruckig]
moveit_ros_visualization: Cannot locate rosdep definition for [ament_lint_common]
moveit_ros_planning: Cannot locate rosdep definition for [ros_testing]
moveit_servo: Cannot locate rosdep definition for [std_srvs]
moveit_task_constructor_msgs: Cannot locate rosdep definition for [rosidl_default_generators]
rviz_marker_tools: Cannot locate rosdep definition for [urdfdom_headers]
moveit_setup_assistant: Cannot locate rosdep definition for [ament_lint_common]
moveit_ros_planning_interface: Cannot locate rosdep definition for [ament_lint_auto]
moveit_ros_warehouse: Cannot locate rosdep definition for [ament_lint_common]
moveit_planners: Cannot locate rosdep definition for [ament_lint_common]
moveit_ros_robot_interaction: Cannot locate rosdep definition for [ament_lint_common]
moveit_planners_ompl: Cannot locate rosdep definition for [ament_lint_common]
moveit_resources_panda_moveit_config: Cannot locate rosdep definition for [robot_state_publisher]
moveit_resources_fanuc_moveit_config: Cannot locate rosdep definition for [robot_state_publisher]
moveit_task_constructor_visualization: Cannot locate rosdep definition for [launch_testing_ros]
pilz_industrial_motion_planner: Cannot locate rosdep definition for [tf2_eigen_kdl]
moveit_common: Cannot locate rosdep definition for [ament_lint_common]
moveit_planners_chomp: Cannot locate rosdep definition for [ament_lint_common]
moveit2_tutorials: Cannot locate rosdep definition for [robot_state_publisher]
moveit_ros_occupancy_map_monitor: Cannot locate rosdep definition for [ament_lint_common]
moveit_resources: Cannot locate rosdep definition for [robot_state_publisher]
moveit_ros_move_group: Cannot locate rosdep definition for [std_srvs]
srdfdom: Cannot locate rosdep definition for [urdfdom_headers]
moveit_runtime: Cannot locate rosdep definition for [ament_lint_common]
moveit_chomp_optimizer_adapter: Cannot locate rosdep definition for [ament_lint_common]
chomp_motion_planner: Cannot locate rosdep definition for [ament_lint_common]
moveit_task_constructor_core: Cannot locate rosdep definition for [launch_testing_ament_cmake]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

```

jomoengineer, 2022-05-26:
This is the output of the two command prior to the rosdep command:
```
~/ws_moveit2/src$ git clone https://github.com/ros-planning/moveit2_tutorials -b main
Cloning into 'moveit2_tutorials'...
remote: Enumerating objects: 94549, done.
remote: Counting objects: 100% (2081/2081), done.
remote: Compressing objects: 100% (1204/1204), done.
remote: Total 94549 (delta 1058), reused 1816 (delta 825), pack-reused 92468
Receiving objects: 100% (94549/94549), 762.71 MiB | 6.02 MiB/s, done.
Resolving deltas: 100% (54011/54011), done.

~/ws_moveit2/src$ vcs import < moveit2_tutorials/moveit2_tutorials.repos
.......
=== ./launch_param_builder (git) ===
Cloning into '.'...
=== ./moveit2 (git) ===
Cloning into '.'...
=== ./moveit_resources (git) ===
Cloning into '.'...
=== ./moveit_task_constructor (git) ===
Cloning into '.'...
=== ./moveit_visual_tools (git) ===
Cloning into '.'...
=== ./rosparam_shortcuts (git) ===
Cloning into '.'...
=== ./srdfdom (git) ===
Cloning into '.'...

```

jomoengineer, 2022-05-27:
Okay, I think I resolved the issue.  It appeared my humble install was not complete or broken.  I ran the 'apt' command again to install humble via binaries and it picked up a bunch of missing packages.
After this, I was able to build MoveIt 2 and the tutorials.

This is the output of the build though:
```
~/ws_moveit2$ colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> srdfdom
Starting >>> geometric_shapes
Starting >>> moveit_common
Starting >>> moveit_resources_panda_description
Finished <<< moveit_common [0.70s]
Starting >>> moveit_resources_fanuc_description
Finished <<< moveit_resources_panda_description [0.96s]
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_fanuc_description [0.75s]
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_panda_moveit_config [0.78s]
Starting >>> moveit_resources_pr2_description
Finished <<< moveit_resources_fanuc_moveit_config [1.11s]
Starting >>> moveit_resources_prbt_support
Finished <<< moveit_resources_pr2_description [2.22s]
Starting >>> moveit_resources
Finished <<< moveit_resources_prbt_support [2.33s]
Finished <<< moveit_resources [3.43s]
Finished <<< srdfdom [20.0s]
Starting >>> moveit_configs_utils
--- stderr: moveit_configs_utils
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< moveit_configs_utils [1.47s]
[Processing: geometric_shapes]
--- stderr: geometric_shapes
In file included from /usr/include/eigen3/Eigen/Core:294,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /home/jomoros/ws_moveit2/src/geometric_shapes/include/geometric_shapes/aabb.h:34,
                 from /home/jomoros/ws_moveit2/src/geometric_shapes/include/geometric_shapes/bodies.h:35,
                 from /home/jomoros/ws_moveit2/src/geometric_shapes/src/bodies.cpp:31:
In member function ‘Eigen::PlainObjectBase<Derived>& Eigen::PlainObjectBase<Derived>::operator=(Eigen::PlainObjectBase<Derived>&&) [with Derived = Eigen::Matrix<double, 3, 1>]’,
    inlined from ‘Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>& Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::operator=(Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>&&) [with _Scalar = double; int _Rows = 3; int _Cols = 1; int _Options = 0; int _MaxRows = 3; int _MaxCols = 1]’ at /usr/include/eigen3/Eigen/src/Core/Matrix.h:281:22,
    inlined from ‘virtual void bodies::ConvexMesh::useDimensions(const shapes::Shape*)’ at /home/jomoros/ws_moveit2/src/geometric_shapes/src/bodies.cpp:852:46:
/usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:504:7: warning: ‘*(Eigen::DenseStorage<double, 3, 3, 1, 0>*)((char*)&<unnamed> + offsetof(Eigen::Vector3d, Eigen::Matrix<double, 3, 1, 0, 3, 1>::<unnamed>.Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >::<unnamed>.Eigen::MatrixBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >::<unnamed>.Eigen::DenseBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >::<unnamed>.Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 3>::<unnamed>.Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 1>::<unnamed>.Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 0>::<unnamed>))’ may be used uninitialized [-Wmaybe-uninitialized]
  504 |       m_storage = std::move(other.m_storage);
      |       ^~~~~~~~~
/home/jomoros/ws_moveit2/src/geometric_shapes/src/bodies.cpp: In member function ‘virtual void bodies::ConvexMesh::useDimensions(const shapes::Shape*)’:
/home/jomoros/ws_moveit2/src/geometric_shapes/src/bodies.cpp:852:46: note: ‘<anonymous>’ declared here
  852 |   mesh_data_->mesh_center_ = Eigen::Vector3d();
      |                                              ^
---
Finished <<< geometric_shapes [56.4s]
Starting >>> moveit_core
[Processing: moveit_core]
[Processing: moveit_core]
[Processing: moveit_core]
[Processing: moveit_core]
[Processing: moveit_core]
[Processing: moveit_core]
[Processing: moveit_core]
[Processing: moveit_core]
Finished <<< moveit_core [4min 10s]
Starting >>> moveit_ros_occupancy_map_monitor
Starting >>> moveit_simple_controller_manager
Starting >>> moveit_resources_prbt_ikfast_manipulator_plugin
Starting >>> chomp_motion_planner
Finished <<< moveit_simple_controller_manager [25.3s]
Starting >>> moveit_plugins
Finished <<< moveit_plugins [2.01s]
Starting >>> pilz_industrial_motion_planner_testutils
Finished <<< moveit_resources_prbt_ikfast_manipulator_plugin [41.8s]
Starting >>> moveit_ros_control_interface
[... 52 lines cut ...]
[Processing: moveit_ros_visualization, moveit_servo, pilz_industrial_motion_planner]
[Processing: moveit_ros_visualization, moveit_servo, pilz_industrial_motion_planner]
[Processing: moveit_ros_visualization, moveit_servo, pilz_industrial_motion_planner]
[Processing: moveit_ros_visualization, moveit_servo, pilz_industrial_motion_planner]
[Processing: moveit_ros_visualization, moveit_servo, pilz_industrial_motion_planner]
[Processing: moveit_ros_visualization, moveit_servo, pilz_industrial_motion_planner]
[Processing: moveit_ros_visualization, moveit_servo, pilz_industrial_motion_planner]
Finished <<< moveit_servo [6min 39s]
Finished <<< pilz_industrial_motion_planner [7min 2s]
Starting >>> moveit_planners
Finished <<< moveit_planners [0.71s]
Starting >>> moveit_runtime
Finished <<< moveit_runtime [0.74s]
[Processing: moveit_ros_visualization]
Finished <<< moveit_ros_visualization [8min 2s]
Starting >>> moveit_ros
Starting >>> moveit_setup_assistant
Finished <<< moveit_ros [0.71s]
Starting >>> moveit
Finished <<< moveit [0.74s]
Finished <<< moveit_setup_assistant [27.9s]

Summary: 41 packages finished [22min 54s]
  2 packages had stderr output: geometric_shapes moveit_configs_utils
```

vatanaksoytezer, 2022-05-27:
Looks good to me! I am closing the issue for now, feel free to open it if you are to have any more issues.

closed 2022-05-27 by vatanaksoytezer

imshaaz21, 2024-03-01:
> Hello, @jomoengineer thank you for reporting an issue. Any chance you missed the rosdep installation step as in: `rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y`?
> 
> Do you mind sharing the output of rosdep?

thanks