moveit/moveit2_tutorials #406: How to use `make multiversion` to build Tutorials

davetcoleman, 2022-06-02:
@vatanaksoytezer [indicates](https://github.com/ros-planning/moveit2_tutorials/pull/400#issuecomment-1145145889) one can build the tutorials locally using ``make multiversion``, but I do not know how.

> It does actually make multiversion and if you need API we have make multiversion-with-api.

The [build_locally](https://github.com/ros-planning/moveit2_tutorials/blob/main/build_locally.sh) script I created long ago does not currently support this. Can someone explain and/or document how to build all the versions of the tutorials at once?

vatanaksoytezer, 2022-06-02:
You can just write `make multiversion` in terminal. Then `xdg-open ./build/html/index.html` to view it.

vatanaksoytezer, 2022-06-02:
Just need to add a few options to build_locally.sh and some documentation. The available options are in https://github.com/ros-planning/moveit2_tutorials/blob/main/Makefile

davetcoleman, 2022-06-02:
Despite double checking that ``pip install sphinx-multiversion`` is installed, which it is via ``build_locally.sh``, I still get the error:

> moveit2_tutorials$ make multiversion
> sphinx-multiversion -c . "." build/html
> /bin/sh: 1: sphinx-multiversion: not found
> make: *** [Makefile:13: multiversion] Error 127

closed 2023-12-01 by tylerjw