moveit/moveit2_tutorials #458: pick and place tutorials can't launch

xlla, 2022-07-02:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Humble]
* OS Version: macOS 10.13.6
* Source or Binary build?
source , mainly base on origin/ros2

### Steps to reproduce
just follow the tutorial of [pick and place](https://moveit.picknik.ai/humble/doc/tutorials/pick_and_place_with_moveit_task_constructor/pick_and_place_with_moveit_task_constructor.html)

in running the demo section, 
it should add install launch folder statements inside CMakeLists.txt, other wise we can't run 
`ros2 launch mtc_tutorial pick_place_demo.launch.py `

after deploy launch folder, it failure to launch.

```
$ ros2 launch mtc_tutorial pick_place_demo.launch.py
[INFO] [launch]: All log files can be found below /Users/xlla/.ros/log/2022-07-03-01-23-42-065169-slim.local-30365
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [mtc_tutorial-1]: process started with pid [30366]
[mtc_tutorial-1] libc++abi.dylib: terminating with uncaught exception of type rclcpp::exceptions::RCLInvalidROSArgsError: failed to parse arguments: Couldn't parse remap rule: '-r __node:=introspection_slim.local_30366_140450084168480'. Error: Expected lexeme type 1, got 22 at index 26, at /Users/xlla/ros2_foxy/src/ros2/rcl/rcl/src/rcl/lexer_lookahead.c:246, at /Users/xlla/ros2_foxy/src/ros2/rcl/rcl/src/rcl/arguments.c:373
[ERROR] [mtc_tutorial-1]: process has died [pid 30366, exit code -6, cmd '/Users/xlla/ws_moveit2/install/lib/mtc_tutorial/mtc_tutorial --ros-args --params-file /var/folders/6j/gqhq_r1n0wg5qjypvtkccnqr0000gn/T/launch_params_zucq2zxa'].
```

even launch file from moveit2_tutorials failure to launch

```
$ ros2 launch moveit2_tutorials pick_place_demo.launch.py
[INFO] [launch]: All log files can be found below /Users/xlla/.ros/log/2022-07-03-01-24-39-186562-slim.local-30408
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [mtc_tutorial-1]: process started with pid [30410]
[mtc_tutorial-1] libc++abi.dylib: terminating with uncaught exception of type rclcpp::exceptions::RCLInvalidROSArgsError: failed to parse arguments: Couldn't parse remap rule: '-r __node:=introspection_slim.local_30410_140550745549520'. Error: Expected lexeme type 1, got 22 at index 26, at /Users/xlla/ros2_foxy/src/ros2/rcl/rcl/src/rcl/lexer_lookahead.c:246, at /Users/xlla/ros2_foxy/src/ros2/rcl/rcl/src/rcl/arguments.c:373
[ERROR] [mtc_tutorial-1]: process has died [pid 30410, exit code -6, cmd '/Users/xlla/ws_moveit2/install/lib/moveit2_tutorials/mtc_tutorial --ros-args --params-file /var/folders/6j/gqhq_r1n0wg5qjypvtkccnqr0000gn/T/launch_params_q2pm6ek6'].

```

other tutorials before "pick and place" have executed successfully.

### Expected behaviour
demo successful launch

### Backtrace or Console output

xlla, 2022-07-03:
I found a solution:
In package moveit_task_constructor_core,  avoid '.' appear in hostname

`std::replace(std::begin(our_hostname), std::end(our_hostname), '.', '_');`

closed 2022-07-03 by xlla

yangming517, 2023-05-19:
hello, I have same question,but I don't find the package moveit_task_constructor_core. where is it ?

rhaschke, 2023-05-19:
@xlla, please file a PR with your solution against the source repository next time. This way others can benefit from the fix as well :wink: