moveit/moveit2_tutorials #479: MoveIt2 tutorials don't work for windows binary install

emielke12, 2022-08-01:
### Description

On the documentation for moveit2 binary install for windows [here](https://moveit.ros.org/install-moveit2/binary-windows/) it states that the moveit2 libraries are included in the install. Then you get a link to the quickstart tutorials [here](https://moveit.picknik.ai/foxy/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html). However, even attempting the first command `ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true` spits back an error indicating the package moveit2_tutorials is not found. 

A quick search of the installed packages lets me know that the moveit packages that are installed include:

- moveit_kinematics
- moveit_ros_benchmarks
- moveit_ros_move_group
- moveit_ros_occupancy_map_monitor
- moveit_ros_planning
- moveit_ros_warehouse

It seems that the package `moveit2_tutorials` is not included in the binary install, and I'll have to install from source myself. 

This would be beneficial to have outlined in the documentation on installation

### Your environment
* ROS Distro: Foxy
* OS Version: Windows 10
* Binary
* If binary, which release version: ros-foxy-desktop v20211109.0.220118.0001

### Steps to reproduce
Install binaries via links shown, and attempt to run command indicated

### Expected behaviour
Failure to find package

### Backtrace or Console output
```

C:\>ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true
Package 'moveit2_tutorials' not found: "package 'moveit2_tutorials' not found, searching: ['C:\\\\opt\\\\ros\\\\foxy\\\\x64']"

```

lukbieli, 2022-08-17:
Same for me. I'm trying to install it for a whole week but everything fails at some point. I've even tried to install it on VirtualBox with ubuntu but it freezes the entire instance during ros installation. Currently, I've abandoned it because it's a waste of time.

vatanaksoytezer, 2022-08-17:
Unfortunately, I don't have any way of testing things on Windows, but everything should work smoothly in Ubuntu, given you have enough resources for your computer. I assume if you can install MoveIt from binaries in Foxy and build the tutorials from source it should still work as expected. The tutorial package is not available as binaries in Ubuntu or Windows, to encourage people to go more in-depth and compile things from source while learning to interact with MoveIt.

closed 2023-12-01 by tylerjw