moveit/moveit2_tutorials #499: libmoveit_robot_interaction.so.2.5.3: undefined reference during CMake Build Process

jmorgan2023, 2022-08-30:
### Description
During the build process of the tutorials, there is an import that is unrecognized in the libmoveit_robot_interaction.so.2.5.3 that prevents successful compilation of the rest of the tutorial suite.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04 (Clean Install)
* Source or Binary build: Binary
* If binary, which release version: 2.5.3-1jamsssssssssssssssssssssssssssssssssssmy.20220808.181146

### Steps to reproduce
Perform a clean install of Ubuntu 22.04
I then followed the initial setup located [here](https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html) to install Ros2 Humble
Then followed the instructions [here](https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html) to attempt to install MoveIt2 and the associated tutorials. 

### Expected behaviour
Build success

### Backtrace or Console output

[View Output Here](https://gist.github.com/jmorgan2023/b03b46d758c86ae20970bb80ba9e233a) or below:

```bash
gridl0ck@gridl0ck-Issued:~/ws_moveit2$ colcon build --mixin release --parallel-workers 1
[0.457s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'controller_manager' is in: /opt/ros/humble
	'hardware_interface' is in: /opt/ros/humble
	'ros2_control_test_assets' is in: /opt/ros/humble
	'controller_interface' is in: /opt/ros/humble
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding controller_interface controller_manager hardware_interface ros2_control_test_assets

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> moveit_common
Finished <<< moveit_common [0.09s]
Starting >>> moveit_resources_panda_description
Finished <<< moveit_resources_panda_description [0.08s]           
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_panda_moveit_config [0.08s]                 
Starting >>> moveit_configs_utils
--- stderr: moveit_configs_utils                    
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< moveit_configs_utils [0.68s]
Starting >>> moveit_resources_fanuc_description
Finished <<< moveit_resources_fanuc_description [0.08s]                   
Starting >>> controller_manager_msgs
Finished <<< controller_manager_msgs [0.64s]                      
Starting >>> ros2_control_test_assets
Finished <<< ros2_control_test_assets [0.09s]                 
Starting >>> hardware_interface
Finished <<< hardware_interface [0.22s]                     
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_fanuc_moveit_config [0.08s]                 
Starting >>> controller_interface
Finished <<< controller_interface [0.17s]                     
Starting >>> moveit_task_constructor_msgs
Finished <<< moveit_task_constructor_msgs [0.58s]                       
Starting >>> rviz_marker_tools
Finished <<< rviz_marker_tools [0.11s]                    
Starting >>> controller_manager
--- stderr: controller_manager                                
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< controller_manager [0.48s]
Starting >>> moveit_resources_pr2_description
Finished <<< moveit_resources_pr2_description [0.09s]                  
Starting >>> moveit_core
Finished <<< moveit_core [0.60s]                      
Starting >>> moveit_ros_occupancy_map_monitor
Finished <<< moveit_ros_occupancy_map_monitor [0.17s]                    
Starting >>> moveit_ros_planning
Finished <<< moveit_ros_planning [0.37s]                      
Starting >>> moveit_kinematics
Finished <<< moveit_kinematics [0.19s]                      
Starting >>> moveit_ros_move_group
Finished <<< moveit_ros_move_group [0.27s]                       
Starting >>> moveit_ros_warehouse
Finished <<< moveit_ros_warehouse [0.20s]                      
Starting >>> moveit_ros_robot_interaction
--- stderr: moveit_ros_robot_interaction                               
/usr/bin/ld: libmoveit_robot_interaction.so.2.5.3: undefined reference to `moveit::core::RobotState::setFromIK(moveit::core::JointModelGroup const*, geometry_msgs::msg::Pose_<std::allocator<void> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, double, std::function<bool (moveit::core::RobotState*, moveit::core::JointModelGroup const*, double const*)> const&, kinematics::KinematicsQueryOptions const&, std::function<double (geometry_msgs::msg::Pose_<std::allocator<void> > const&, moveit::core::RobotState const&, moveit::core::JointModelGroup*, std::vector<double, std::allocator<double> > const&)>)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/locked_robot_state_test.dir/build.make:329: locked_robot_state_test] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:185: CMakeFiles/locked_robot_state_test.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_ros_robot_interaction [0.40s, exited with code 2]

Summary: 20 packages finished [6.02s]
  1 package failed: moveit_ros_robot_interaction
  3 packages had stderr output: controller_manager moveit_configs_utils moveit_ros_robot_interaction
  42 packages not processed
```

vatanaksoytezer, 2022-08-30:
I think you have might have mixed binaries and sources. Can you try to delete MoveIt binaries if you have any `sudo apt remove ros-humble-moveit*`?

jmorgan2023, 2022-08-30:
After doing that and running the command again I am running into dependency issues with earlier builds, 

```bash
colcon build --mixin release
[0.467s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'controller_interface' is in: /opt/ros/humble
	'controller_manager' is in: /opt/ros/humble
	'hardware_interface' is in: /opt/ros/humble
	'ros2_control_test_assets' is in: /opt/ros/humble
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding controller_interface controller_manager hardware_interface ros2_control_test_assets

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> moveit_common
Starting >>> moveit_resources_panda_description
Starting >>> moveit_configs_utils
Starting >>> moveit_resources_fanuc_description
Starting >>> controller_manager_msgs
Starting >>> ros2_control_test_assets
Starting >>> moveit_task_constructor_msgs
Starting >>> rviz_marker_tools
Finished <<< moveit_resources_panda_description [0.42s]              
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_common [0.48s]
Starting >>> moveit_resources_pr2_description
Finished <<< moveit_resources_fanuc_description [0.51s]             
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< ros2_control_test_assets [0.52s]
Starting >>> hardware_interface
Finished <<< rviz_marker_tools [0.55s]
Starting >>> moveit_resources_prbt_support
Finished <<< moveit_resources_panda_moveit_config [0.29s]           
Starting >>> joint_limits
Finished <<< moveit_resources_pr2_description [0.35s]               
Starting >>> moveit_core
Finished <<< moveit_resources_fanuc_moveit_config [0.33s]
Finished <<< moveit_resources_prbt_support [0.28s]
Starting >>> rviz_visual_tools
Starting >>> rosparam_shortcuts
--- stderr: moveit_task_constructor_msgs                            
CMake Error at CMakeLists.txt:6 (find_package):
  By not providing "Findmoveit_msgs.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "moveit_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "moveit_msgs" with
  any of the following names:

    moveit_msgsConfig.cmake
    moveit_msgs-config.cmake

  Add the installation prefix of "moveit_msgs" to CMAKE_PREFIX_PATH or set
  "moveit_msgs_DIR" to a directory containing one of the above files.  If
  "moveit_msgs" provides a separate development package or SDK, be sure it
  has been installed.

gmake: *** [Makefile:2907: cmake_check_build_system] Error 1
---
Failed   <<< moveit_task_constructor_msgs [0.89s, exited with code 2]
Aborted  <<< joint_limits [0.21s]
Aborted  <<< rviz_visual_tools [0.06s]
Aborted  <<< rosparam_shortcuts [0.05s]
Aborted  <<< hardware_interface [0.39s]
Aborted  <<< moveit_core [0.10s]
Aborted  <<< controller_manager_msgs [1.00s]      
Aborted  <<< moveit_configs_utils [1.16s]                       

Summary: 9 packages finished [1.52s]
  1 package failed: moveit_task_constructor_msgs
  7 packages aborted: controller_manager_msgs hardware_interface joint_limits moveit_configs_utils moveit_core rosparam_shortcuts rviz_visual_tools
  2 packages had stderr output: moveit_configs_utils moveit_task_constructor_msgs
  46 packages not processed
```

vatanaksoytezer, 2022-08-30:
You should re-run `sudo apt update && rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y` under your src directory.

jmorgan2023, 2022-08-30:
Ok! That worked!

Summary of resolution:

When working with the same system as above, do NOT execute the following command ` install ros-humble-moveit`. Instead, follow the instructions on how to install Ros2 from [here](https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html).

Then, follow this guide located [here](https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html) to install MoveIt2!

closed 2022-08-30 by jmorgan2023