moveit/moveit2_tutorials #502: The Robotarm Take No Reaction after Following the Step in "Visualizing MoveIt in RViz"

Sgt-Harry, 2022-09-01:
### Description
I was following the step in Visualizing MoveIt in RViz. Everything seems good before the second click of the next botton. Rviz gives me the info that it has successfully planed the path. But when I try to press it again to execute the plan, it just have no reaction. I had the same problem on VM before, when I was thinking it might be the issue of VM. But it still happened after I changed my platform to WSL2. 

### Your environment
* ROS Distro: [Humble]
* OS Version: e.g. Ubuntu 22.04
* Source build
* source build by ''git clone https://github.com/ros-planning/moveit2_tutorials -b main --depth 1''

### Steps to reproduce
I followd the step in tutorial,. 

### Expected behaviour
The Robotarm in Rviz should execute the plan and draw a line.

### Backtrace or Console output
output by hello_moveit
```
harry@LAPTOP-9979S1EP:~/ws_moveit2$ ros2 run hello_moveit hello_moveit
[INFO] [1662024801.002254457] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.180845 seconds
[INFO] [1662024801.002593745] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[WARN] [1662024801.049706096] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
[INFO] [1662024801.076990685] [move_group_interface]: Ready to take commands for planning group panda_arm.
[INFO] [1662024801.080549867] [hello_moveit.remote_control]: RemoteControl Ready.
[INFO] [1662024801.080855020] [hello_moveit.remote_control]: Waiting to continue: Press 'Next' in the RvizVisualToolsGui window to plan
[INFO] [1662024858.834484878] [hello_moveit.remote_control]: ... continuing
[INFO] [1662024858.837432860] [move_group_interface]: MoveGroup action client/server ready
```
output by rviz
```harry@LAPTOP-9979S1EP:~/ws_moveit2$ ros2 launch moveit2_tutorials demo.launch.py
[INFO] [launch]: All log files can be found below /home/harry/.ros/log/2022-09-01-17-26-35-984776-LAPTOP-9979S1EP-18608
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [18609]
[INFO] [static_transform_publisher-2]: process started with pid [18611]
[INFO] [robot_state_publisher-3]: process started with pid [18613]
[INFO] [move_group-4]: process started with pid [18615]
[INFO] [ros2_control_node-5]: process started with pid [18617]
[INFO] [spawner-6]: process started with pid [18619]
[INFO] [spawner-7]: process started with pid [18621]
[INFO] [spawner-8]: process started with pid [18623]
[static_transform_publisher-2] [WARN] [1662024396.408955800] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-2] [INFO] [1662024396.434717335] [static_transform_publisher]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'world' to 'panda_link0'
[robot_state_publisher-3] Link panda_link1 had 1 children
[robot_state_publisher-3] Link panda_link2 had 1 children
[robot_state_publisher-3] Link panda_link3 had 1 children
[robot_state_publisher-3] Link panda_link4 had 1 children
[robot_state_publisher-3] Link panda_link5 had 1 children
[robot_state_publisher-3] Link panda_link6 had 1 children
[robot_state_publisher-3] Link panda_link7 had 1 children
[robot_state_publisher-3] Link panda_link8 had 1 children
[robot_state_publisher-3] Link panda_hand had 2 children
[robot_state_publisher-3] Link panda_leftfinger had 0 children
[robot_state_publisher-3] Link panda_rightfinger had 0 children
[robot_state_publisher-3] [INFO] [1662024396.446044875] [robot_state_publisher]: got segment panda_hand
[robot_state_publisher-3] [INFO] [1662024396.446240232] [robot_state_publisher]: got segment panda_leftfinger
[robot_state_publisher-3] [INFO] [1662024396.446258396] [robot_state_publisher]: got segment panda_link0
[robot_state_publisher-3] [INFO] [1662024396.446263816] [robot_state_publisher]: got segment panda_link1
[robot_state_publisher-3] [INFO] [1662024396.446267784] [robot_state_publisher]: got segment panda_link2
[robot_state_publisher-3] [INFO] [1662024396.446271551] [robot_state_publisher]: got segment panda_link3
[robot_state_publisher-3] [INFO] [1662024396.446275088] [robot_state_publisher]: got segment panda_link4
[robot_state_publisher-3] [INFO] [1662024396.446278694] [robot_state_publisher]: got segment panda_link5
[robot_state_publisher-3] [INFO] [1662024396.446282451] [robot_state_publisher]: got segment panda_link6
[robot_state_publisher-3] [INFO] [1662024396.446286168] [robot_state_publisher]: got segment panda_link7
[robot_state_publisher-3] [INFO] [1662024396.446289905] [robot_state_publisher]: got segment panda_link8
[robot_state_publisher-3] [INFO] [1662024396.446293592] [robot_state_publisher]: got segment panda_rightfinger
[ros2_control_node-5] [INFO] [1662024396.457092476] [resource_manager]: Loading hardware 'PandaFakeSystem' 
[ros2_control_node-5] [INFO] [1662024396.459162297] [resource_manager]: Initialize hardware 'PandaFakeSystem' 
[... 444 lines cut ...]
  moveit::planning_interface::MoveGroupInterface::Plan msg;
  auto const ok = static_cast<bool>(move_group_interface.plan(msg));
  return std::make_pair(ok, msg);
}();

// Execute the plan
if (success) {
  draw_trajectory_tool_path(plan.trajectory_);
  moveit_visual_tools.trigger();
  prompt("Press 'Next' in the RvizVisualToolsGui window to execute");
  draw_title("Executing");
  moveit_visual_tools.trigger();
  move_group_interface.execute(plan);
} else {
  draw_title("Planning Failed!");
  moveit_visual_tools.trigger();
  RCLCPP_ERROR(logger, "Planing failed!");
}

  // Shutdown ROS
  rclcpp::shutdown();
  spinner.join();  // <--- Join the thread before exiting
  return 0;
}
```

tylerjw, 2022-09-01:
Thank you for your bug report. I'll try to reproduce this myself and see if I can figure it out.

AndyZe, 2022-09-01:
It's almost certainly this bug. Still.

https://github.com/ros-planning/moveit2/issues/1474

tylerjw, 2022-09-01:
Can you try using CycloneDDS or following the advice in that issue?

Sgt-Harry, 2022-09-01:
> Can you try using CycloneDDS or following the advice in that issue?

Yes! After I installed the CycloneDDS und moved the MGI initialization before SingleThreadedExecutor. It works perfecty!
It has solved the problem that bothered me for a long time. Really appreciate that.
Maybe we need some changes in the tutorial document.

closed 2022-09-01 by Sgt-Harry

tylerjw, 2022-09-02:
@Sgt-Harry would you like to make a PR to add a note to the tutorials about this issue and what changes worked for addressing it for you?

AndyZe, 2022-09-02:
How about we fix it? As posted in the maintainer notes from last meeting -- https://discourse.ros.org/t/moveit-maintainer-meeting-aug-25/27090/3

A dedicated callback group in moveit_visual_tools might do it

tylerjw, 2022-09-02:
My understanding from that meeting was that this was fixed upstream in FastDDS but we are waiting on the sync to have the fix available.

Sgt-Harry, 2022-09-06:
> @Sgt-Harry would you like to make a PR to add a note to the tutorials about this issue and what changes worked for addressing it for you?

I'm glad to do that!
Its kind of a honor for me actually.

yangming517, 2023-04-24:
hello ,would you please describe carefully how you use CycloneDDS  to solve this problem? I'm a beginner  of ros2.Hope your reply,thanks

yangming517, 2023-04-25:
I have installed CycloneDDS  by 'sudo apt install ros-humble-rmw-cyclonedds-cpp' ,and added 'export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp' into my /.bashrc,but it still don't work