moveit/moveit2_tutorials #506: Tutorial - Won't start rviz out of the box

MightyMirko, 2022-09-05:
### Description
After following all tutorials this line 
ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true
won't work as predicted 
`move_group-4] [ERROR] [1662375043.529176769] [moveit_robot_model.robot_model]: Joint 'wheel_left_joint' not found in model 'panda'
[move_group-4] [ERROR] [1662375043.529211279] [moveit_robot_model.robot_model]: Joint 'wheel_right_joint' not found in model 'panda'
`
### Your environment
* ROS Distro: Galactic
* OS Version: e.g. Ubuntu 20.04
* git clone -b galactic https://github.com/ros-planning/moveit2_tutorials.git --depth 1

### Steps to reproduce
Freshly install, ros2 via ros2 tutorial and moveit2 via tutorial - 
https://moveit.picknik.ai/galactic/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html#getting-started

### Expected behaviour
rviz should open with panda robot

### Backtrace or Console output

[Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.](https://gist.githubusercontent.com/MightyMirko/c371c035d11626f21605bf18e4b848a3/raw/d6a511abf2d2b8b43a5c51b2d361bfdacf8765d6/error.log)

rhaschke, 2022-09-05:
Looks like you loaded the wrong URDF. Wheels are definitely not part of the Panda robot...
Check whether you have a robot_description parameter set from some previous ROS run.

MightyMirko, 2022-09-05:
How would one check that? 

Nonetheless, a reboot helped...

rhaschke, 2022-09-05:
```
rosparam list
rosparam get robot_description
```

closed 2022-09-05 by rhaschke