moveit/moveit2_tutorials #508: Pandabot won't show until i set export LC_NUMERIC="en_US.UTF-8"

MightyMirko, 2022-09-07:
Description

After following all tutorials this line
ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true
won't work as predicted
move_group-4] [ERROR] [1662375043.529176769] [moveit_robot_model.robot_model]: Joint 'wheel_left_joint' not found in model 'panda' [move_group-4] [ERROR] [1662375043.529211279] [moveit_robot_model.robot_model]: Joint 'wheel_right_joint' not found in model 'panda'
Your environment

    ROS Distro: Galactic
    OS Version: e.g. Ubuntu 20.04
    git clone -b galactic https://github.com/ros-planning/moveit2_tutorials.git --depth 1

Steps to reproduce

Freshly install, ros2 via ros2 tutorial and moveit2 via tutorial -
https://moveit.picknik.ai/galactic/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html#getting-started
Expected behaviour

rviz should open with panda robot
Backtrace or Console output

[Use [gist.github.com](https://github.com/ros-planning/moveit2_tutorials/issues/gist.github.com) to copy-paste the console output or segfault backtrace using gdb.](https://gist.githubusercontent.com/MightyMirko/c371c035d11626f21605bf18e4b848a3/raw/d6a511abf2d2b8b43a5c51b2d361bfdacf8765d6/error.log)

---------

i need to set 

    export LC_NUMERIC="en_US.UTF-8"

then it will work

MightyMirko, 2022-09-07:
`rviz2-1] [ERROR] [1662547064.760901552] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': parameter 'panda_arm.kinematics_solver_timeout' has invalid type: Wrong parameter type, parameter {panda_arm.kinematics_solver_timeout} is of type {double}, setting it to {string} is not allowed.
`

MightyMirko, 2022-09-07:
isn't it an issue by having , instead of . in decimal numbers?

rhaschke, 2022-09-07:
Yes, locale=de_DE expects a comma instead of a dot due do a bug in parameter parsing. However, it's not a bug related to liburdfdom, which was another well-known issue in the past.

MightyMirko, 2022-09-07:
And to answer your question: yes! the exception is gone after setting to en_US

tylerjw, 2023-12-01:
This is a known issue with ROS. The instructions for installing ROS (https://docs.ros.org/en/rolling/Installation/Ubuntu-Install-Debians.html#set-locale) include a section on setting your locale for this reason.

closed 2023-12-01 by tylerjw