moveit/moveit2_tutorials #511: moveit_ros_control_interface failed with colcon build --mixin release

enriLoniterp, 2022-09-14:
### Description

I'm trying to do the basic tutorial of Moveit2 humble. I have followed all the procedures of the Getting Started but `moveit_ros_control_interface` ifailed

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source binary
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html

Uninstalled ROS 2 Humble, deleted every dir created also in opt.
Reinstalled, and get to this tutorial

### Expected behaviour
It should compile with colcon build --mixin release, instead it gave me error on a packace

### Backtrace or Console output
```
rosdev@rosdev:~/ws_moveit2/src$ source /opt/ros/humble/setup.bash
rosdev@rosdev:~/ws_moveit2/src$ colcon build --mixin release
[0.466s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'hardware_interface' is in: /opt/ros/humble
	'controller_interface' is in: /opt/ros/humble
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding controller_interface hardware_interface

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> moveit_common
Starting >>> moveit_resources_panda_description
Starting >>> moveit_configs_utils
Starting >>> moveit_resources_fanuc_description
Starting >>> controller_manager_msgs
Starting >>> ros2_control_test_assets
Starting >>> moveit_task_constructor_msgs
Starting >>> rviz_marker_tools
Finished <<< moveit_common [1.09s]                                   
Starting >>> moveit_resources_pr2_description
Finished <<< moveit_configs_utils [1.10s]
Finished <<< moveit_resources_panda_description [1.12s]
Starting >>> moveit_resources_prbt_support
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_fanuc_description [1.18s]
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< ros2_control_test_assets [1.82s]                                
Starting >>> hardware_interface
Finished <<< moveit_resources_prbt_support [0.87s]                           
Starting >>> joint_limits
Finished <<< moveit_resources_panda_moveit_config [0.91s]
Finished <<< moveit_resources_pr2_description [0.96s]
Starting >>> rviz_visual_tools
Starting >>> moveit_core
Finished <<< moveit_resources_fanuc_moveit_config [0.88s]
Starting >>> rosparam_shortcuts
[Processing: controller_manager_msgs, hardware_interface, joint_limits, moveit_core, moveit_task_constructor_msgs, rosparam_shortcuts, rviz_marker_tools, rviz_visual_tools]
Finished <<< rviz_marker_tools [39.7s] 
Starting >>> moveit_resources
Finished <<< joint_limits [37.8s]       
Finished <<< controller_manager_msgs [40.1s]
Finished <<< moveit_resources [0.91s]   
Finished <<< moveit_task_constructor_msgs [45.5s]
Finished <<< rosparam_shortcuts [45.8s]                                        
[Processing: hardware_interface, moveit_core, rviz_visual_tools]               
Finished <<< rviz_visual_tools [1min 22s]    
Finished <<< hardware_interface [1min 26s]   
Starting >>> controller_interface
Starting >>> transmission_interface
[Processing: controller_interface, moveit_core, transmission_interface]
Finished <<< controller_interface [37.1s]    
Starting >>> controller_manager
Finished <<< transmission_interface [47.0s]  
[Processing: controller_manager, moveit_core]
Finished <<< controller_manager [1min 4s]    
Starting >>> ros2controlcli
Finished <<< ros2controlcli [1.08s]          
Starting >>> ros2_control
Finished <<< ros2_control [0.86s]            
[Processing: moveit_core]                                         
[Processing: moveit_core]                                         
Finished <<< moveit_core [4min 29s]                                
Starting >>> moveit_ros_occupancy_map_monitor
Starting >>> moveit_simple_controller_manager
Starting >>> moveit_resources_prbt_ikfast_manipulator_plugin
Starting >>> chomp_motion_planner            
Starting >>> pilz_industrial_motion_planner_testutils
Finished <<< moveit_simple_controller_manager [22.1s]
Starting >>> moveit_plugins                  
Starting >>> moveit_ros_control_interface
Finished <<< moveit_plugins [0.88s]          
Finished <<< pilz_industrial_motion_planner_testutils [27.1s]
Finished <<< chomp_motion_planner [34.9s]    
Starting >>> moveit_chomp_optimizer_adapter
Starting >>> moveit_planners_chomp
Finished <<< moveit_resources_prbt_ikfast_manipulator_plugin [38.1s]
--- stderr: moveit_ros_control_interface    
/home/rosdev/ws_moveit2/src/moveit2/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp: In member function ‘bool moveit_ros_control_interface::MoveItControllerManager::fixChainedControllers(std::shared_ptr<controller_manager_msgs::srv::ListControllers_Response_<std::allocator<void> > >&)’:
/home/rosdev/ws_moveit2/src/moveit2/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp:471:22: error: ‘struct controller_manager_msgs::msg::ControllerState_<std::allocator<void> >’ has no member named ‘chain_connections’
  471 |       if (controller.chain_connections.size() > 1)
      |                      ^~~~~~~~~~~~~~~~~
/home/rosdev/ws_moveit2/src/moveit2/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp:478:56: error: ‘struct controller_manager_msgs::msg::ControllerState_<std::allocator<void> >’ has no member named ‘chain_connections’
  478 |       for (const auto& chained_controller : controller.chain_connections)
      |                                                        ^~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/moveit_ros_control_interface_plugin.dir/build.make:76: CMakeFiles/moveit_ros_control_interface_plugin.dir/src/controller_manager_plugin.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:143: CMakeFiles/moveit_ros_control_interface_plugin.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_ros_control_interface [16.3s, exited with code 2]
Aborted  <<< moveit_planners_chomp [3.68s]             
Aborted  <<< moveit_chomp_optimizer_adapter [3.75s]    
Aborted  <<< moveit_ros_occupancy_map_monitor [39.0s]

Summary: 28 packages finished [5min 10s]
  1 package failed: moveit_ros_control_interface
  3 packages aborted: moveit_chomp_optimizer_adapter moveit_planners_chomp moveit_ros_occupancy_map_monitor
  1 package had stderr output: moveit_ros_control_interface
  31 packages not processed

```

enriLoniterp, 2022-09-14:
please someone tells me what's wrong, it's been a day i'm waisting time trying to solve this bug but i think is something above me.

Please, someone help me!

enriLoniterp, 2022-09-14:
non funziona una minchia!!!

AndyZe, 2022-09-14:
Please be polite.

I think the cause was this commit:  https://github.com/ros-planning/moveit2/pull/1482

Maybe the change it depends on in ros2_control was not released for Humble yet.

As a temporary fix, while waiting, you could roll the state of `moveit2` back to prior to that commit.

```
cd moveit2
git reset --hard 337416de7cbfbb58f151f16edd4f8c0f36e5faf8
```

AndyZe, 2022-09-14:
You can also make sure your other ros2 packages are up-to-date.

`sudo apt update`
`sudo apt upgrade`

enriLoniterp, 2022-09-14:
Hi @AndyZe ,
i've seen that build farm is completely right, but i uninstalled ROS2 (also every file produced) and installed from scratch, same process adopted with moveit2 from the tutorial and still it doesn't work.

Also updated and upgraded my version using apt.

i can't figure it out why it's giving me errors in a .cpp file inside code. It tells one variable is missing of a parameter. I don't think i should manually change something by myself but also i'm confused on what to do. Also remake everything from scratch again.
Thank you for your help!

Enrico

enriLoniterp, 2022-09-15:
> Also, note that the buildfarm does not have any problem with Humble. This is an indication that the issue is on your end, I'm sorry to say:
> 
> https://github.com/ros-planning/moveit2#buildfarm
Hi @AndyZe,
i've just created a fresh built ubuntu 22.04 virtual machine running on VMWare, installed all packages with `sudo apt update && sudo apt upgrade`, i've installed everything from scratch: ROS2 Humble source following with attention the official guide http://docs.ros.org.ros.informatik.uni-freiburg.de/en/humble/Installation/Ubuntu-Install-Debians.html and then proceeded with Moveit installation from https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html.
Followed every passage described here but it gave me same error as before in the same with same package. Maybe i have something missing on my basic setup, but i'm not sure to have many problems about it.

Thank you again, Enrico

BrettHemes, 2022-09-16:
I had the same issue awhile back.  Per the "Getting Started" instructions, vcs import is run with `https://github.com/ros-planning/moveit2_tutorials/blob/main/moveit2_tutorials.repos` which checks out version `2.12.1` of `ros2_control` which has an older `ControllerState` msg (i.e., doesn't have `chain_connections` field).

Checking out `2.14.0` (i.e., `humble`) should get the updated message description and fix the issue.  Perhaps the vcs repo file wasn't updated with https://github.com/ros-planning/moveit2/pull/1482 that Andy mentioned.

```
$ cd ws_moveit/src/ros2_control
$ git fetch origin humble
$ git checkout humble
$ cd ../..
$ colcon build --mixin release
```

enriLoniterp, 2022-09-16:
hi @BrettHemes,
thank you very much for the help.

Now i've made git reset hard as mentioned by @AndyZe  and worked fine (for my purpose), i will try it next days and confirm your suggestions.

Thanks a lot for your help,
Enrico

lorepieri8, 2022-09-21:
I'm getting the same behaviour using the Docker install (https://moveit.picknik.ai/humble/doc/how_to_guides/how_to_setup_docker_containers_in_ubuntu.html), but I solved following BrettHemes instructions.

closed 2023-12-01 by tylerjw