moveit/moveit2_tutorials #512: 404 in docker tutorial

lorepieri8, 2022-09-16:
### Description

When following https://moveit.picknik.ai/main/doc/how_to_guides/how_to_setup_docker_containers_in_ubuntu.html
 and trying to download the config file:

wget https://raw.githubusercontent.com/ros-planning/moveit2_tutorials/how-to-docker-ubuntu/_scripts/docker-compose.yml

### Your environment
* ROS Distro: Rolling
* OS Version: e.g. Ubuntu 20.04

### Backtrace or Console output

```
-2022-09-16 13:02:07--  https://raw.githubusercontent.com/ros-planning/moveit2_tutorials/how-to-docker-ubuntu/_scripts/docker-compose.yml
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.110.133, 185.199.109.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2022-09-16 13:02:08 ERROR 404: Not Found.
```

vatanaksoytezer, 2022-09-16:
@lorepieri8 thank you for noticing. The link should be changed with: https://raw.githubusercontent.com/ros-planning/moveit2_tutorials/main/_scripts/docker-compose.yml. Do you mind opening a quick PR if you want to contribute? Else I'll replace it soon.

sjahr, 2023-12-01:
Fixed with PR above.

closed 2023-12-01 by sjahr