moveit/moveit2_tutorials #518: Cannot launch MoveIt Quickstart in RViz

lorepieri8, 2022-09-21:
### Description

Cannot launch MoveIt Quickstart in RViz

### Your environment
* ROS Distro: Humble
* OS Version: e.g. Ubuntu 22.04
* Source or Binary build: Docker install (https://moveit.picknik.ai/humble/doc/how_to_guides/how_to_setup_docker_containers_in_ubuntu.html)

### Steps to reproduce

ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true

### Backtrace or Console output

[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: 'robot_description_planning'

stephanie-eng, 2022-09-28:
This I think was an issue with MoveIt Configs Utils attempting to load some parameters related to Pilz that did not apply in the tutorial. This should have been fixed in #520 and #521, which should now properly load Pilz. 

Can you try again with the latest MoveIt Tutorials and confirm if this issue is resolved or not?

sjahr, 2023-12-01:
Seems to be outdated now. Please re-open if it is still the case

closed 2023-12-01 by sjahr