moveit/moveit2_tutorials #525: Not understand MoveItGroupInterface

enriLoniterp, 2022-10-04:
### Description

I've downloaded Universalrobots driver and followed the procedure to use moveIt!2, it actually works what showed in the README and i use "ur_manipulator" in Rviz to move my arm. But my interest is to use moveit2_tutorials to send pose and design a pick and place application with MoveItGriuoInterface. 

What i have designed is:

```
#include <memory>

#include <rclcpp/rclcpp.hpp>
#include <moveit/move_group_interface/move_group_interface.h>

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);
  auto const node = std::make_shared<rclcpp::Node>(
    "hello_moveit",
    rclcpp::NodeOptions().automatically_declare_parameters_from_overrides(true)
  );

  auto const logger = rclcpp::get_logger("ur_manipulator");

  // Create the MoveIt MoveGroup Interface
using moveit::planning_interface::MoveGroupInterface;
auto move_group_interface = MoveGroupInterface(node, "ur_manipulator");
printf("picvkandplace");

// Set a target Pose
auto const target_pose = []{
  geometry_msgs::msg::Pose msg;
  msg.orientation.w = 1.0;
  msg.position.x = 0.28;
  msg.position.y = -0.2;
  msg.position.z = 0.5;
  return msg;
}();
move_group_interface.setPoseTarget(target_pose);

// Create a plan to that target pose
auto const [success, plan] = [&move_group_interface]{
  moveit::planning_interface::MoveGroupInterface::Plan msg;
  auto const ok = static_cast<bool>(move_group_interface.plan(msg));
  return std::make_pair(ok, msg);
}();

// Execute the plan
if(success) {
  move_group_interface.execute(plan);
} else {
  RCLCPP_ERROR(logger, "Planing failed!");
}
  // Shutdown ROS
  rclcpp::shutdown();
  return 0;
}
```

i launch ros2 run hello_moveit hello_moveit as in the tutorial but i get:
```
[ERROR] [1664871447.339399807] [hello_moveit]: Could not find parameter robot_description_semantic and did not receive robot_description_semantic via std_msgs::msg::String subscription within 10.000000 seconds.
Error:   Could not parse the SRDF XML File. Error=XML_ERROR_EMPTY_DOCUMENT ErrorID=13 (0xd) Line number=0
         at line 715 in ./src/model.cpp
[ERROR] [1664871447.342434566] [moveit_rdf_loader.rdf_loader]: Unable to parse SRDF
[FATAL] [1664871447.342740966] [move_group_interface]: Unable to construct robot model. Please make sure all needed information is on the parameter server.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to construct robot model. Please make sure all needed information is on the parameter server.
[ros2run]: Aborted
```
I can't understand what should i look for also because in that everything is set to use moveit, `robot_description_semantic included`.

Hope someone could help to this endless problem!

Enrico

jwson97, 2022-10-05:
I have same problem. Who can solve this problem?

felipebapa, 2023-03-23:
@enriLoniterp Did you find an answer? I am developing a similar program. 
Thanks for your help!

okvik, 2023-03-23:
You are likely missing the step of launching the `move_group` node for your robot package before running the program. 

For example, you can try `ros2 launch moveit_resources_panda_moveit_config demo.launch.py`. If you are using a different robot there likely alredy exists a package and a launch file that does the same, otherwise you're gonna need to either 1) use MoveIt Setup Assistant to generate an appropriate `moveit_config` package, or 2) write the launch file by hand.

All of this is more or less covered in the MoveIt docs.

okvik, 2023-03-23:
The problem is that your program (actually the MoveGroupInterface constructor) cannot find the SRDF robot description `MoveGroupInterface` in either of the two places it looks in: 1) in the `robot_description_semantic` parameter in your node's parameters list, or 2) in a published `/robot_description_semantic` ROS topic.

It follows that you can solve this in two ways: either 1) feed the `robot_description_semantic` parameter to your program node when launching it, or 2) configure the `move_group` node to publish its own `robot_description_semantic` parameter to a matching topic from which other nodes in the system can read it.

Personally I prefer the second method. 

To do this, find the launch file that is creating the `move_group` node and set its node parameter `publish_robot_description_semantic = True` . I suggest you set the `publish_robot_description` as well.

To verify the SRDF is published correctly run

```
ros2 topic echo /robot_description_semantic
```

Which should spit out the `.srdf` file.

felipebapa, 2023-03-23:
Thanks for your fast answer!. I am going to check.

sjahr, 2023-12-01:
Looks like the discussion is resolved. Feel free to re-open if you'd like to continue it

closed 2023-12-01 by sjahr