moveit/moveit2_tutorials #528: Issue with robot_description_semantic in running the tutorial "Your first C++ MoveIt Project"

Sraobot, 2022-10-07:
### Description and Steps to reproduce
I am trying to run the [Your First C++ MoveIt Project](https://moveit.picknik.ai/humble/doc/tutorials/your_first_project/your_first_project.html). The code I used is [hello_moveit.cpp](https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/tutorials/your_first_project/hello_moveit.cpp) which is provided at the end of tutorial webpage. When I run `ros2 run hello_moveit hello_moveit`, I get this error 
``` 
[ERROR] [1664720561.886608829] [hello_moveit]: Could not find parameter robot_description_semantic and did not receive robot_description_semantic via std_msgs::msg::String subscription within 10.000000 seconds. 
Error:   Could not parse the SRDF XML File. Error=XML_ERROR_EMPTY_DOCUMENT ErrorID=13 (0xd) Line number=0
         at line 715 in ./src/model.cpp
[ERROR] [1664720561.900546627] [moveit_rdf_loader.rdf_loader]: Unable to parse SRDF
[FATAL] [1664720561.901184806] [move_group_interface]: Unable to construct robot model. Please make sure all needed information is on the parameter server.
    terminate called after throwing an instance of 'std::runtime_error'
    what():  Unable to construct robot model. Please make sure all needed information is on the parameter server.
[ros2run]: Aborted 
```
Before running the command `ros2 run hello_moveit hello_moveit`, I did run `ros2 launch moveit2_tutorials demo.launch.py` as mentioned in the tutorial.

The error mentions could not find `robot_description_semantic` but when I run the command `ros2 param get | grep robot_description_semantic`, I get 6 results from the nodes `/move_group, /moveit_simple_controller_manager` and `/rviz2`. Each of the mentioned nodes has 2 parameters `robot_description_semantic` and `publish_description_semantic` which makes 6 parameters as reported by `grep`. 

I try to check what is in `robot_description_semantic` for each of the nodes using `ros2 param get <node_name> robot_description_semantic`, I get the same result for each of them which is probably a URDF file. 

Now, I check for `publish_robot_description_semantic` in each of those nodes. 
For `/move_group` and `/moveit_simple_controller_manager`, I get ```Boolean Value is: False``` and for `/rviz2` I get `Parameter not set`.
In this comment, https://github.com/ros-planning/moveit2_tutorials/pull/262#issuecomment-1014927379 it was mentioned about setting `"publish_robot_description_semantic"` to `True`. So, I try to do this.

For `/move_group` node, I set `publish_robot_description_semantic` parameter to True but still not able to run `ros2 run hello_moveit hello_moveit`. Then, I tried to do for `/rviz2`, it gets set successfully to True but still the code does not run. Then, I try for `/moveit_simple_controller_manager`, the parameter change command returns `Setting parameter failed`. 

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source build as mentioned in the tutorial

### Expected behaviour
`ros2 run hello_moveit hello_moveit` should run and move the arm to the desired pose as mentioned in hello_moveit.cpp.

tylerjw, 2022-10-07:
Thank you for the bug report. I'll try to reproduce this to see if I can figure out what is broken.

tylerjw, 2022-10-07:
Sweet, I think I know what happened. That looks like you checked out the rolling version of the tutorials and not the humble version. That means that you also built a workspace that is setup for rolling and not humble.

[image]

Please delete the code in your source directory and start over on this starting with the Download Source step of getting started:

- https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html#download-source-code-of-moveit-and-the-tutorials

Sraobot, 2022-10-07:
Ok. I will build again and let you know.

Sraobot, 2022-10-08:
Thanks. It works now.

closed 2022-10-08 by Sraobot

pirate999, 2023-04-02:
I have the same problem, I use   
Ubuntu20.04, ros foxy, moveit2
and I git clone moveit2_tutorials foxy branch

> ERROR] [1680406587.092033426] [hello_moveit]: Could not find parameter robot_description_semantic and did not receive robot_description_semantic via std_msgs::msg::String subscription within 10.000000 seconds

pirate999, 2023-04-02:
> I have the same problem, I use Ubuntu20.04, ros foxy, moveit2 and I git clone moveit2_tutorials foxy branch

Hi, I have the same problem, I use
Ubuntu20.04, ros foxy, moveit2
and I git clone moveit2_tutorials foxy branch

ERROR] [1680406587.092033426] [hello_moveit]: Could not find parameter robot_description_semantic and did not receive robot_description_semantic via std_msgs::msg::String subscription within 10.000000 seconds

pirate999, 2023-04-02:
> Sweet, I think I know what happened. That looks like you checked out the rolling version of the tutorials and not the humble version. That means that you also built a workspace that is setup for [...]

Hi, I have the same problem, I use
Ubuntu20.04, ros foxy, moveit2
and I git clone moveit2_tutorials foxy branch

yinguoxiangyi, 2023-05-29:
I use the galactic branch and have also the same problem.
[answers.ros.org#407290](https://answers.ros.org/question/407290/issue-in-moveit2-tutorial-regarding-robot_description_semantic-parameter) and https://github.com/ros-planning/moveit2_tutorials/issues/631 give the hint.
Adding following lines to `planning_scene_monitor_parameters` dictionary in file `demo.launch.py` of package `moveit2_tutorials` can solve the issue.
```
 "publish_robot_description":True,
 "publish_robot_description_semantic":True,
```

tylerjw, 2023-05-30:
We removed the galactic and foxy versions of the tutorials from the site because there are issues with them; they are EOL versions of ROS and are not supported.

RodBelaFarin, 2023-06-15:
Same problem with humble, too.

newtonjeri, 2024-03-22:
> I use the galactic branch and have also the same problem. [answers.ros.org#407290](https://answers.ros.org/question/407290/issue-in-moveit2-tutorial-regarding-robot_description_semantic-parameter) [...]

Thanks, it worked for me, am using ROS2 humble

navdotnetreqs, 2024-08-24:
> Thanks, it worked for me, am using ROS2 humble

Worked for me too, ROS2 Iron

zainkarsan7, 2024-09-17:
i have the same issue, it looks like robot_description and robot_description_semantic are missing and the traceback from rviz shows similar to above, 

`[rviz2-4] [ERROR] [1726584574.262224199] [rviz]: Could not find parameter robot_description_semantic and did not receive robot_description_semantic via std_msgs::msg::String subscription within 10.000000 seconds.
[rviz2-4] Error:   Could not parse the SRDF XML File. Error=XML_ERROR_EMPTY_DOCUMENT ErrorID=13 (0xd) Line number=0
[rviz2-4]          at line 732 in /root/ws_moveit2/src/srdfdom/src/model.cpp
[rviz2-4] [ERROR] [1726584574.300945614] [moveit_rdf_loader.rdf_loader]: Unable to parse SRDF
[rviz2-4] [ERROR] [1726584574.328735327] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Robot model not loaded`

ros2 param get /rviz robot_description_semantic
Parameter not set.

I am trying to load a ur10e config from moveit setup assistant. I already changed my doubles to ints, and some other capabilities setting according to other forum posts on the internet, but don't know how to fix this. I can confirm that the moveit2 tutorials all work fine for me. Any help to get this running would be great, thanks