moveit/moveit2_tutorials #537: Building Colcon Workspace

jamaalskeete, 2022-10-31:
### **Description**
Not able to build colcon workspace following Moveit2 humble instructions

### **Your environment**
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* BInary Build (Debian)

### Steps to reproduce
Followed the instructions in the following link after installing ros2 (desktop) several times
https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html

### **Expected behaviour**
Build colcon workspace without any errors

### **Actual Behavior**
<pre>Summary: 30 packages finished [19min 13s]
  1 package failed: moveit_planners_ompl
  2 packages aborted: moveit_ros_move_group moveit_visual_tools
  3 packages had stderr output: launch_param_builder moveit_configs_utils moveit_planners_ompl
  22 packages not processed
</pre>

### **Console output**

<pre>--- stderr: moveit_planners_ompl              
<b>/home/davinci/ws_moveit2/src/moveit2/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp:</b> In member function ‘<b>int32_t ompl_interface::ModelBasedPlanningContext::logPlannerStatus(ompl::geometric::SimpleSetupPtr)</b>’:
<b>/home/davinci/ws_moveit2/src/moveit2/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp:983:52:</b> <font color="#C01C28"><b>error: </b></font>‘<b>START_STATE_INVALID</b>’ is not a member of ‘<b>moveit_msgs::msg::MoveItErrorCodes</b>’ {aka ‘<b>moveit_msgs::msg::MoveItErrorCodes_&lt;std::allocator&lt;void&gt; &gt;</b>’}
  983 |       result = moveit_msgs::msg::MoveItErrorCodes::<font color="#C01C28"><b>START_STATE_INVALID</b></font>;
      |                                                    <font color="#C01C28"><b>^~~~~~~~~~~~~~~~~~~</b></font>
<b>/home/davinci/ws_moveit2/src/moveit2/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp:987:52:</b> <font color="#C01C28"><b>error: </b></font>‘<b>GOAL_STATE_INVALID</b>’ is not a member of ‘<b>moveit_msgs::msg::MoveItErrorCodes</b>’ {aka ‘<b>moveit_msgs::msg::MoveItErrorCodes_&lt;std::allocator&lt;void&gt; &gt;</b>’}
  987 |       result = moveit_msgs::msg::MoveItErrorCodes::<font color="#C01C28"><b>GOAL_STATE_INVALID</b></font>;
      |                                                    <font color="#C01C28"><b>^~~~~~~~~~~~~~~~~~</b></font>
<b>/home/davinci/ws_moveit2/src/moveit2/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp:991:52:</b> <font color="#C01C28"><b>error: </b></font>‘<b>UNRECOGNIZED_GOAL_TYPE</b>’ is not a member of ‘<b>moveit_msgs::msg::MoveItErrorCodes</b>’ {aka ‘<b>moveit_msgs::msg::MoveItErrorCodes_&lt;std::allocator&lt;void&gt; &gt;</b>’}
  991 |       result = moveit_msgs::msg::MoveItErrorCodes::<font color="#C01C28"><b>UNRECOGNIZED_GOAL_TYPE</b></font>;
      |                                                    <font color="#C01C28"><b>^~~~~~~~~~~~~~~~~~~~~~</b></font>
<b>/home/davinci/ws_moveit2/src/moveit2/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp:1006:52:</b> <font color="#C01C28"><b>error: </b></font>‘<b>CRASH</b>’ is not a member of ‘<b>moveit_msgs::msg::MoveItErrorCodes</b>’ {aka ‘<b>moveit_msgs::msg::MoveItErrorCodes_&lt;std::allocator&lt;void&gt; &gt;</b>’}
 1006 |       result = moveit_msgs::msg::MoveItErrorCodes::<font color="#C01C28"><b>CRASH</b></font>;
      |                                                    <font color="#C01C28"><b>^~~~~</b></font>
<b>/home/davinci/ws_moveit2/src/moveit2/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp:1010:52:</b> <font color="#C01C28"><b>error: </b></font>‘<b>ABORT</b>’ is not a member of ‘<b>moveit_msgs::msg::MoveItErrorCodes</b>’ {aka ‘<b>moveit_msgs::msg::MoveItErrorCodes_&lt;std::allocator&lt;void&gt; &gt;</b>’}
 1010 |       result = moveit_msgs::msg::MoveItErrorCodes::<font color="#C01C28"><b>ABORT</b></font>;
      |                                                    <font color="#C01C28"><b>^~~~~</b></font>
gmake[2]: *** [ompl_interface/CMakeFiles/moveit_ompl_interface.dir/build.make:104: ompl_interface/CMakeFiles/moveit_ompl_interface.dir/src/model_based_planning_context.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:186: ompl_interface/CMakeFiles/moveit_ompl_interface.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   &lt;&lt;&lt; moveit_planners_ompl [5min 40s, exited with code 2]
Aborted  &lt;&lt;&lt; moveit_visual_tools [1min 44s]              
Aborted  &lt;&lt;&lt; moveit_ros_move_group [3min 51s]

Summary: 30 packages finished [19min 13s]
  1 package failed: moveit_planners_ompl
  2 packages aborted: moveit_ros_move_group moveit_visual_tools
  3 packages had stderr output: launch_param_builder moveit_configs_utils moveit_planners_ompl
  22 packages not processed
</pre>

Abishalini, 2022-10-31:
Make sure you have `moveit_msgs` cloned into your workspace. Pull down all the latest changes in the `ros2` branch.
If you have package debians installed, make sure to uninstall it `sudo apt remove --purge ros-humble-moveit-msgs`

jamaalskeete, 2022-10-31:
Hey thanks for replying so quickly. So two questions for you. Why am I cloning moveit_msgs only to then remove it? How do I pull down all the latest change in the ros2 branch?

Abishalini, 2022-10-31:
You would be cloning in your workspace
```
cd ~/ws_moveit2/src
git clone --branch ros2 https://github.com/ros-planning/moveit_msgs.git 
cd moveit_msgs
git pull 
```
and removing the `moveit_msgs` installed in `/opt/ros/humble` by executing `sudo apt remove --purge ros-humble-moveit-msgs`

AndyZe, 2022-10-31:
> Hey thanks for replying so quickly. So two questions for you. Why am I cloning moveit_msgs only to then remove it? How do I pull down all the latest change in the ros2 branch?

I don't believe the instructions ever say to do `sudo apt install ros-humble-moveit-msgs`. If they do, that's a mistake and we should fix it.

jamaalskeete, 2022-10-31:
> You would be cloning in your workspace
> 
> ```
> cd ~/ws_moveit2/src
> git clone --branch ros2 https://github.com/ros-planning/moveit_msgs.git 
> cd moveit_msgs
> git pull 
> ```
> 
> and removing the `moveit_msgs` installed in `/opt/ros/humble` by executing `sudo apt remove --purge ros-humble-moveit-msgs`
I did this and the entire process completed. The console outputs this now:

<pre>Summary: 56 packages finished [51min 7s]
  3 packages had stderr output: launch_param_builder moveit_configs_utils moveit_setup_srdf_plugins</pre>

Thank you for your help

closed 2022-10-31 by AndyZe