moveit/moveit2_tutorials #541: cmake permission error when following Getting Started tutorial

fschiro, 2022-11-06:
### Description
When working through the [Getting Started tutorial](https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html) I'm getting an error at the section "Building your Colcon Workspace:" 

```
--- stderr: srdfdom                                               
CMake Error: failed to create symbolic link 'libsrdfdom.so': Operation not permitted
CMake Error: cmake_symlink_library: System Error: Operation not permitted
gmake[2]: *** [CMakeFiles/srdfdom.dir/build.make:127: libsrdfdom.so.2.0.4] Error 1
gmake[2]: *** Deleting file 'libsrdfdom.so.2.0.4'
gmake[1]: *** [CMakeFiles/Makefile2:162: CMakeFiles/srdfdom.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
```
This is some kind of permission error with CMake. After researching I'm unsure how to fix it. 

Command that produced error: ```colcon build --mixin release```
### Your environment
* ROS Distro: Humbe
* OS Version: Ubuntu 22.04
* Source or Binary build? Unsure tutorial did not make it clear
* 
### Steps to reproduce
I followed steps in tutorial. 

### Expected behaviour
No permissions error

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

gavanderhoorn, 2022-11-06:
do you have you workspace on a file system that's placed on a USB stick, an external HDD/SDD or a network drive (NFS)? Or is it a partition 'shared' with Windows?

FAT, FAT32, ExFat and NTFS file systems don't support all the required features for Linux to correctly manage files and their permission, which could lead to the errors you mention.

Could also be a permissions problem for your specific user, or the way the partition is mounted. Or a problem with specific directories and ACLs (often seen in university setups with student accounts on shared infrastructure).

This is very likely not a problem with MoveIt, nor ROS.

henningkayser, 2023-05-16:
closing due to inactivity. This really doesn't seem to be related to the build setup at all. @fschiro please reopen and comment if this problem still exists for you.