moveit/moveit2_tutorials #544: Launch exception TypeError demo.launch.py

quarkytale, 2022-11-10:
### Description

`demo.launch.py` throwing exception and shutdown while launching on a fresh installation.
Trace in short:
```
  File "/home/spaceros-user/moveit2/install/moveit_configs_utils/lib/python3.8/site-packages/moveit_configs_utils/moveit_configs_builder.py", line 87, in <module>
    @dataclass(slots=True)
TypeError: dataclass() got an unexpected keyword argument 'slots'

```

### Environment
* ROS Distro: Humble
* OS Version: Ubuntu 20.04
* Source build: `moveit_core` and `moveit2_tutorials` on humble, other deps generated from `rosinstall_generator` or on `ros2` branch

### Steps to reproduce
`ros2 launch -d moveit2_tutorials demo.launch.py` on a fresh moveit2 installation.
Can share my Dockerfile if required.

### Expected behaviour
Running the demo without any errors, rviz showing up with the arm.

### Backtrace or Console output

```
user@osrf-Predator-PT715-51:~/moveit2$ ros2 launch moveit2_tutorials demo.launch.py
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2022-11-10-06-53-25-981433-osrf-Predator-PT715-51-72
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: dataclass() got an unexpected keyword argument 'slots'
user@osrf-Predator-PT715-51:~/moveit2$ ros2 launch -d moveit2_tutorials demo.launch.py
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2022-11-10-07-28-53-945579-osrf-Predator-PT715-51-74
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f45f19f1130>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f45f19f1130>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7f45f19f1070>'
Executing <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /home/user/ros/build/launch/launch/launch_service.py:228> exception=InvalidLaunchFileError('py') created at /home/user/ros/build/launch/launch/launch_service.py:318> took 0.203 seconds
[DEBUG] [launch]: An exception was raised in an async action/event
[DEBUG] [launch]: Traceback (most recent call last):
  File "/home/user/ros/build/launch/launch/launch_description_sources/any_launch_file_utilities.py", line 54, in get_launch_description_from_any_launch_file
    return loader(launch_file_path)
  File "/home/user/ros/build/launch/launch/launch_description_sources/python_launch_file_utilities.py", line 62, in get_launch_description_from_python_launch_file
    launch_file_module = load_python_launch_file_as_module(python_launch_file_path)
  File "/home/user/ros/build/launch/launch/launch_description_sources/python_launch_file_utilities.py", line 37, in load_python_launch_file_as_module
    loader.exec_module(mod)
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/user/moveit2/install/moveit2_tutorials/share/moveit2_tutorials/launch/demo.launch.py", line 9, in <module>
    from moveit_configs_utils import MoveItConfigsBuilder
  File "/home/user/moveit2/install/moveit_configs_utils/lib/python3.8/site-packages/moveit_configs_utils/__init__.py", line 1, in <module>
    from .moveit_configs_builder import MoveItConfigsBuilder, MoveItConfigs
  File "/home/user/moveit2/install/moveit_configs_utils/lib/python3.8/site-packages/moveit_configs_utils/moveit_configs_builder.py", line 87, in <module>
    @dataclass(slots=True)
TypeError: dataclass() got an unexpected keyword argument 'slots'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/user/ros/build/launch/launch/launch_service.py", line 336, in run_async
    raise completed_tasks_exceptions[0]
  File "/home/user/ros/build/launch/launch/launch_service.py", line 230, in _process_one_event
    await self.__process_event(next_event)
  File "/home/user/ros/build/launch/launch/launch_service.py", line 250, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/home/user/ros/build/launch/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/home/user/ros/build/launch/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/home/user/ros/build/launch/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/home/user/ros/build/launch/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/home/user/ros/build/launch/launch/actions/include_launch_description.py", line 148, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/home/user/ros/build/launch/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/home/user/ros/build/launch/launch/launch_description_sources/any_launch_description_source.py", line 53, in _get_launch_description
    return get_launch_description_from_any_launch_file(location)
  File "/home/user/ros/build/launch/launch/launch_description_sources/any_launch_file_utilities.py", line 57, in get_launch_description_from_any_launch_file
    raise InvalidLaunchFileError(extension, likely_errors=exceptions)
launch.invalid_launch_file_error.InvalidLaunchFileError: Caught exception when trying to load file of format [py]: dataclass() got an unexpected keyword argument 'slots'

[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: dataclass() got an unexpected keyword argument 'slots'
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.Shutdown'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f45f19f1fa0>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f45f19f1fa0>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f45f19f15e0>'
```

JafarAbdi, 2022-11-10:
I don't think it's recommended, we officially only support tier 1 platforms https://docs.ros.org/en/foxy/Releases/Release-Humble-Hawksbill.html

quarkytale, 2022-11-10:
I'm planning to port to jammy but please let me know if there's any workaround for making things work in focal as well.

JafarAbdi, 2022-11-10:
I think reverting https://github.com/ros-planning/moveit2/pull/1308 should resolve that issue

quarkytale, 2022-11-18:
Moved to Ubuntu Jammy!

closed 2022-11-18 by quarkytale