moveit/moveit2_tutorials #546: cannot built moveit2 tutorials

fschiro, 2022-11-13:
### Description

Cant build moveit2 tutorial from [Getting Started](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html) guide

### Your environment
* ROS Distro: Rolling
* OS Version: Ubuntu 22.04
* Source or Binary build: I believe source but just following steps in the guide

### Steps to reproduce

```
source /opt/ros/rolling/setup.bash  
sudo apt install python3-rosdep

sudo rosdep init
rosdep update
sudo apt update
sudo apt dist-upgrade

sudo apt install python3-colcon-common-extensions
sudo apt install python3-colcon-mixin
colcon mixin add default https://raw.githubusercontent.com/colcon/colcon-mixin-repository/master/index.yaml
colcon mixin update default

sudo apt install python3-vcstool

mkdir -p ~/ws_moveit/src
cd ~/ws_moveit/src
git clone https://github.com/ros-planning/moveit2_tutorials -b main --depth 1

vcs import < moveit2_tutorials/moveit2_tutorials.repos
sudo apt update && rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
cd ~/ws_moveit
colcon build --mixin release debug --parallel-workers 1
```

### Expected behaviour
successful build

### Backtrace or Console output

```
[0.358s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'controller_manager' is in: /opt/ros/rolling
	'hardware_interface' is in: /opt/ros/rolling
	'ros2_control_test_assets' is in: /opt/ros/rolling
	'controller_interface' is in: /opt/ros/rolling
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding controller_interface controller_manager hardware_interface ros2_control_test_assets

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> moveit_common
Finished <<< moveit_common [0.20s]                   
Starting >>> moveit_resources_panda_description
Finished <<< moveit_resources_panda_description [0.17s]                   
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_panda_moveit_config [0.18s]                   
Starting >>> moveit_configs_utils
--- stderr: moveit_configs_utils                    
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< moveit_configs_utils [0.73s]
Starting >>> moveit_resources_fanuc_description
Finished <<< moveit_resources_fanuc_description [0.16s]                   
Starting >>> controller_manager_msgs
[... 250 lines cut ...]
  111 |     case moveit::core::MoveItErrorCode::GOAL_STATE_INVALID:
      |                                         ^~~~~~~~~~~~~~~~~~
/home/frank/ws_moveit/src/moveit2/moveit_core/utils/include/moveit/utils/moveit_error_code.h:113:41: error: ‘UNRECOGNIZED_GOAL_TYPE’ is not a member of ‘moveit::core::MoveItErrorCode’
  113 |     case moveit::core::MoveItErrorCode::UNRECOGNIZED_GOAL_TYPE:
      |                                         ^~~~~~~~~~~~~~~~~~~~~~
/home/frank/ws_moveit/src/moveit2/moveit_core/utils/include/moveit/utils/moveit_error_code.h:135:41: error: ‘CRASH’ is not a member of ‘moveit::core::MoveItErrorCode’
  135 |     case moveit::core::MoveItErrorCode::CRASH:
      |                                         ^~~~~
/home/frank/ws_moveit/src/moveit2/moveit_core/utils/include/moveit/utils/moveit_error_code.h:137:41: error: ‘ABORT’ is not a member of ‘moveit::core::MoveItErrorCode’
  137 |     case moveit::core::MoveItErrorCode::ABORT:
      |                                         ^~~~~
gmake[2]: *** [planning_interface/CMakeFiles/moveit_planning_interface.dir/build.make:90: planning_interface/CMakeFiles/moveit_planning_interface.dir/src/planning_response.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[2]: *** [planning_interface/CMakeFiles/moveit_planning_interface.dir/build.make:76: planning_interface/CMakeFiles/moveit_planning_interface.dir/src/planning_interface.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1325: planning_interface/CMakeFiles/moveit_planning_interface.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_core [1min 44s, exited with code 2]

Summary: 14 packages finished [3min 50s]
  1 package failed: moveit_core
  3 packages had stderr output: controller_manager moveit_configs_utils moveit_core
  48 packages not processed
```

AndyZe, 2022-11-13:
The first thing to check is to uninstall any moveit debians:  `sudo apt remove ros-rolling-moveit*`

Now that you've prompted me, I'll make a PR to add it to the instructions. Thanks for that.

AndyZe, 2022-11-13:
PR here:  https://github.com/ros-planning/moveit.ros.org/pull/763

henningkayser, 2023-05-16:
@fschiro looks like an issue with your workspace related to ros2_control, @AndyZe's points should have helped you out. Can you comment on that?

closed 2023-12-01 by tylerjw