moveit/moveit2_tutorials #547: Trying to replicate pick and place tutorial in ROS2 Humble with UR

marioesteban11, 2022-11-14:
### Description

Hi everyone, Iḿ trying to replicate [this tutorial](https://moveit.picknik.ai/humble/doc/tutorials/pick_and_place_with_moveit_task_constructor/pick_and_place_with_moveit_task_constructor.html), but Iḿ not able to do it with the UR3e robot.

### Your environment
* ROS Distro: Humble
* OS Version:  Ubuntu 22.04

I yhink I understand the code with the panda robot and how it works, but I'm not able to do it. I think my problem is here:

` const auto& arm_group_name = "panda_arm";

  const auto& hand_group_name = "hand";

  const auto& hand_frame = "panda_hand";
`

This is panda's code. But in UR I think I dont have any "hand" that it seem to be the end effector where u can set the Inverse Kinematics positions and much more.

Do you have any idea that can help me?

Thanks in advance

eyal-friedman, 2022-11-22:
Hi
I have the exact same issues..

Any news?

marioesteban11, 2022-11-29:
Nope, not any, do you? @eyal-friedman

henningkayser, 2023-05-16:
@marioesteban11 I don't understand what's wrong. Can you please elaborate what behavior you are experiencing, ideally with error logs or a reproducible example? And yes, if you want to replicate examples on another robot, you should make sure that the semantics are fully defined in the srdf. In your case, solving for grasps with a "hand" group can't work if it doesn't exist (see `stage_open_hand` which relies on a group). The tutorial is also meant as an example for showing how MTC works in an application. If you only have an arm without end-effector, it would be best to learn MTC building up from only a small sequence of stages.

Wellington-Noberto, 2023-06-14:
I think we're all having the same issue. I'll try to explain the problem that @marioesteban11 is having.

Following the tutorial, we define only 3 properties for the task, namely: `"group"`, `"eef"`, and `"ik_frame"`. However, according to the following error message, we should also define a `"hand"` property:
```
[mtc_tutorial-1] [INFO] [1686779764.410241100] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[mtc_tutorial-1] terminate called after throwing an instance of 'moveit::task_constructor::Property::undeclared'
[mtc_tutorial-1]   what():  Property 'hand': undeclared
```
Accordingly, those 4 properties are listed in the following error log:
```
[mtc_tutorial-1]         149:   // In fact, `task` itself is a SerialContainer by default.
[mtc_tutorial-1]         150:   {
[mtc_tutorial-1]         151:     auto grasp = std::make_unique<mtc::SerialContainer>("pick object");
[mtc_tutorial-1]       > 152:     task.properties().exposeTo(grasp->properties(), { "eef", "hand", "group", "ik_frame" });
[mtc_tutorial-1]         153:     // clang-format off
[mtc_tutorial-1]         154:     grasp->properties().configureInitFrom(mtc::Stage::PARENT,
[mtc_tutorial-1]         155:                                           { "eef", "hand", "group", "ik_frame" });
```

Running the `pick_and_place` launch file we can visualize in the Rviz that the cylinder object is correctly spawned, but the task is not executed.

Fawad433, 2024-08-13:
This is what I have understood. 
- Go to your robot.srdf and find the groups in it 

This is how my file looks like 
```
<group name="myrobot_arm">
        <joint name="arm_base_joint"/>
        <joint name="base_link1_joint"/>
        <joint name="link1_link2_joint"/>
        <joint name="link2_link3_joint"/>
        <joint name="link3_link4_joint"/>
        <joint name="link4_link5_joint"/>
        <joint name="link5_hand_joint"/>
    </group>
    <group name="myrobot_hand">
        <link name="hand"/>
        <link name="hand_leftfinger"/>
        <link name="hand_rightfinger"/>
```

So according to this my lines look like this 

const auto& arm_group_name = "myrobot_arm";
  const auto& hand_group_name = "myrobot_hand";
  const auto& hand_frame = "hand"; // This is the fixed joint of the hand 

I hope this will resolve your issue.