moveit/moveit2_tutorials #549: cant build tutorial in moveit2 humble docker container

fschiro, 2022-11-20:
### Description

The tutorial [MoveIt Quickstart in RViz](https://moveit.picknik.ai/humble/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html#moveit-quickstart-in-rviz) does not seem to work in the MoveIt2 Docker container.   

I built the moveit-2 humble docker container following the guide [How to Set Up MoveIt 2 Docker Containers in Ubuntu](https://moveit.picknik.ai/humble/doc/how_to_guides/how_to_setup_docker_containers_in_ubuntu.html#how-to-set-up-moveit-2-docker-containers-in-ubuntu)](https://moveit.picknik.ai/humble/doc/how_to_guides/how_to_setup_docker_containers_in_ubuntu.html). 

I then tried to build the tutorial and some of the packages failed to build successfully and the tutorial moveit2 demo would not work. 

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source or Binary build? container default or source

### Steps to reproduce
  
Install and start docker container following tutorial steps:   
```
wget https://raw.githubusercontent.com/abake48/moveit2_tutorials/how-to-docker-ubuntu/_scripts/start-docker.sh -O ~/.local/bin/start-docker.sh
chmod +x ~/.local/bin/start-docker.sh

start-docker.sh moveit2-humble moveit/moveit2:humble-source false
start-docker.sh moveit2-humble
source install/setup.bash
```
  
Install and build tutorial package following steps in guide. Below, the difference for version 1 and 2 is the inclusion or exclusion of the tutorial step ```vcs import < moveit2_tutorials/moveit2_tutorials.repos```  which I believe rebuilds moveit2 from source. 
 
**Version 1**
In this version, I  skipped the step ```vcs import < moveit2_tutorials/moveit2_tutorials.repos``` because I assume the docker container already has this built.  So these are the commands I entered:

```
cd ../
mkdir -p ~/ws_moveit2/src
cd ~/ws_moveit2/src
git clone https://github.com/ros-planning/moveit2_tutorials -b humble --depth 1
sudo apt update && rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
cd ~/ws_moveit2
colcon build --mixin release
```

**Version 2:** 
In this version I included ```vcs import < moveit2_tutorials/moveit2_tutorials.repos``` . In this version two packages failed to build and I believe this caused the motion planning tutorial to not work correctly. 

```
cd ../
mkdir -p ~/ws_moveit2/src
cd ~/ws_moveit2/src
git clone https://github.com/ros-planning/moveit2_tutorials -b humble --depth 1
vcs import < moveit2_tutorials/moveit2_tutorials.repos
sudo apt update && rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
cd ~/ws_moveit2
colcon build --mixin release
```

These are the two failed packages:
```
Summary: 55 packages finished [10.0s]
  2 packages had stderr output: launch_param_builder moveit_configs_utils
```
I launched RViz with this command: ```ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true```  but I could not see the two interactive markers. Also motion plan and execute did not seem to work. 

 

### Expected behaviour

The tutorial package should build and the demo MoveIt Quickstart in RViz should work. 

### Backtrace or Console output

**version 2:** 
[... 114 lines cut ...]
  By not providing "Findmoveit_task_constructor_core.cmake" in
  CMAKE_MODULE_PATH this project has asked CMake to find a package
  configuration file provided by "moveit_task_constructor_core", but CMake
  did not find one.

  Could not find a package configuration file provided by
  "moveit_task_constructor_core" with any of the following names:

    moveit_task_constructor_coreConfig.cmake
    moveit_task_constructor_core-config.cmake

  Add the installation prefix of "moveit_task_constructor_core" to
  CMAKE_PREFIX_PATH or set "moveit_task_constructor_core_DIR" to a directory
  containing one of the above files.  If "moveit_task_constructor_core"
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   <<< moveit2_tutorials [3.20s, exited with code 1]

Summary: 0 packages finished [3.30s]
  1 package failed: moveit2_tutorials
  1 package had stderr output: moveit2_tutorials

```

fschiro, 2022-11-22:
The [main](https://moveit.picknik.ai/main/doc/how_to_guides/how_to_setup_docker_containers_in_ubuntu.html) branch docker install tutorial used docker-compose and this worked.  
The [Humble](https://moveit.picknik.ai/humble/doc/how_to_guides/how_to_setup_docker_containers_in_ubuntu.html#how-to-set-up-moveit-2-docker-containers-in-ubuntu) branch link did not work.

I used this command to start the tutorial:
```
DOCKER_IMAGE=rolling-tutorial docker-compose run --rm --name moveit2_container cpu
```
Since ```rolling``` can be changed to ```humble``` maybe the humble guide should be changed to be like the main branch guide.

closed 2022-11-22 by fschiro

stenwire, 2023-12-31:
I happened to have this error while building on Humble, how do I solve it?
@fschiro

stenwire, 2023-12-31:
> I happened to have this error while building on Humble, how do I solve it? @fschiro
> 
> My error:
> 
> ```cmake
>   By not providing "Findmoveit_task_constructor_core.cmake" in
>   CMAKE_MODULE_PATH this project has asked CMake to find a package
>   configuration file provided by "moveit_task_constructor_core", but CMake
>   did not find one.
> 
>   Could not find a package configuration file provided by
>   "moveit_task_constructor_core" with any of the following names:
> 
>     moveit_task_constructor_coreConfig.cmake
>     moveit_task_constructor_core-config.cmake
> 
>   Add the installation prefix of "moveit_task_constructor_core" to
>   CMAKE_PREFIX_PATH or set "moveit_task_constructor_core_DIR" to a directory
>   containing one of the above files.  If "moveit_task_constructor_core"
>   provides a separate development package or SDK, be sure it has been
>   installed
> ```
> 
> After running: ` colcon build --mixin release`, from the docs `https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html`

Fixed

raihan619, 2025-05-03:
How did u fix this issue, im not using a docker, im trying to run https://github.com/PickNikRobotics/UR10e_welding_demo
because im having trouble doing hybrid planning in my roobt, i have followed demo tutorial using panda arm from moveit2 but im stuck in making the planning scene work, and also using OMPL+PRMstakCOnfig planner work. as i dont find a way to use them in the tutorial. 
so i was hoping to get insight from this uR10e welding demo, but im stuck at this phase.