moveit/moveit2_tutorials #570: How to add the position constrain to the planning algorithm in humble

yaqianchen, 2023-01-10:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Humble]
* OS Version: e.g. Ubuntu 22.04
* binary built

I am new to ros2 learning, I want to add the position constraint to my own robot configuration on its end effector. The ompl_constrained_planning in the moveit2_tutorial humble branch seems to be based on ros1. I tried to add the setup.py and setup.cfg to make the python file executable and use the ros2 run to run the node but failed.  I wonder how to run the ompl_constrained_planning file in humble or if there are other recommended ways to add position constraints to my robot. Thanks a lot for any help and suggestions.

akhilkp777, 2023-02-09:
moveit_cpp_tutorial-4] [INFO] [1675937187.657785499] [moveit_ros.current_state_monitor]: Didn't received robot state (joint angles) with recent timestamp within 1.000000 seconds.
[moveit_cpp_tutorial-4] Check clock synchronization if your are running ROS across multiple machines!
[moveit_cpp_tutorial-4] [ERROR] [1675937187.657812738] [move_group_interface]: Failed to fetch current robot state

I got the above error when the box constraint is added in ros2 cpp

henningkayser, 2023-05-17:
@yaqianchen does the tutorial work with a  more recent version of the tutorials? There have been some issues in the past, but in general it should work. The Python launch file is not based on ROS 1 and should not require additional setup steps.

tylerjw, 2023-12-01:
Here is the constraint tutorial: https://moveit.picknik.ai/main/doc/how_to_guides/using_ompl_constrained_planning/ompl_constrained_planning.html

closed 2023-12-01 by tylerjw