moveit/moveit2_tutorials #581: add a description to the docs of how to use moveit with other robots than the panda

FlorianPix, 2023-01-24:
### Description

The tutorials always use the panda arm but it is not directly stated how to use moveit2 with a different robot.

AndyZe, 2023-01-24:
That would be a nice tutorial to have!

FlorianPix, 2023-02-07:
@AndyZe I'd love to create an example for the UR10e. As you have also worked on that project could you maybe help me get started ?

AndyZe, 2023-02-07:
Yeah, I can try to outline something for you this weekend. If that sounds OK.

FlorianPix, 2023-02-10:
Sounds great, either you write an outline or we have an online meeting. You can reach me on discord (Florian Pix#5905) or via e-mail (florian.pix@mailbox.tu-dresden.de).

sshaizkhan, 2023-02-24:
@AndyZe @FlorianPix 
Recently I have created moveit_config for UR5 and was able to create a mtc tutorial similar to panda handa for ur5. Please let me know if I can contribute in any way. Thanks

henningkayser, 2023-05-17:
Did anything come out of this? It would be great to support other robots and we have had related discussions recently. Tagging @tylerjw @abake48 @moriarty for possible collaboration

sshaizkhan, 2023-05-24:
@henningkayser 
Hi Henning,
Let me know if I can be of any help or contribute to this development. Thanks

closed 2023-12-01 by tylerjw