moveit/moveit2_tutorials #582: hello_moveit should warn of limitations and reference MGI for advanced use cases

henningkayser, 2023-01-25:
[hello_moveit ](https://moveit.picknik.ai/humble/doc/tutorials/your_first_project/your_first_project.html) is great for getting something running, but it seems like some users take it as a starting point for doing actual development. Of course, nothing besides very basic planning is even supported. Most features in MoveIt will simply not work since proper launch parameters are required for full MGI functionality.

Here is an example issue that highlights the problem https://github.com/ros-planning/moveit2/issues/1843.

We should warn about the limitations in the tutorial and directly point users at the MGI tutorial or others for further development.

grace-lovelace, 2023-01-27:
Before opening a small PR to address this, would a solution to the issue just be an update to the hello_moveit documentation with a note near the top of the page addressing the limited scope of the initial tutorial and a link to the MGI tutorial page?

henningkayser, 2023-02-27:
@grace-lovelace yes, that would be all that's needed for the start. Thanks for picking this up!

sameergupta4873, 2023-04-02:
@henningkayser if no one is working on this issue, I have made changes to hello_moveit tutorials and issued a PR.

closed 2023-05-17 by henningkayser