moveit/moveit2_tutorials #586: MoveIt Quickstart in RViz does not launch rviz2

zflat, 2023-02-09:
### Description

When running the demo.launch.py file as instructed at https://moveit.picknik.ai/humble/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html the following command does not launch rviz 

```
ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true` 
```

However changing `demo.launch.py rviz_tutorial:=true` does launch RViz, for example with
```
ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=false
```

This is due to a change made in https://github.com/ros-planning/moveit2_tutorials/commit/e292e4da088420c41269e5301be132a1ca9661ca#diff-a3efd55c9d8991e9a4a31bd3652364beaf91be99fd35307b50f8aa95b077a50c

The fix is to add `rviz_node_tutorial` back to the `nodes_to_start` array in the file `doc/tutorials/quickstart_in_rviz/launch/demo.launch.py`.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source or Binary build? Source
* If binary, which release version?
* If source, which git commit or tag?
```
~/ws_moveit2/src/moveit2_tutorials$ git log -1
commit e292e4da088420c41269e5301be132a1ca9661ca (grafted, HEAD -> humble, origin/humble)
Author: mergify[bot] <37929162+mergify[bot]@users.noreply.github.com>
Date:   Thu Jan 26 15:10:12 2023 -0500
```

### Steps to reproduce

* Set up as instructed at https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html
* Run
  ```
   ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true
   ```
* RViz does not launch

### Expected behaviour
RViz should be launched.

airbutt, 2023-02-11:
Took me a day to find that out. You made my day ! by the way binary is not working at all for me, had to build it completely. In Binary there seems some packages missing. But this is another story.....

anujay28, 2023-02-12:
Yeah , I faced a similar issue. Did you clone the `main` repo  and then build it using `colcon` ?

Albert-lee-7, 2023-02-14:
I add rviz_node_tutorial to the nodes_to_start array like this
[image]
but still cannot launch rviz, am I add it wrong or there are other lines to be added.

Albert-lee-7, 2023-02-14:
> I add rviz_node_tutorial to the nodes_to_start array like this [image] but still cannot launch rviz, am I add it wrong or there are other lines to be added.

I rebuild it, and RViz launched as expected

MikeWrock, 2023-03-03:
@zflat I don't have a bare metal humble install but the [`humble-tutorial`](https://github.com/ros-planning/moveit2_tutorials/pkgs/container/moveit2_tutorials/73231979?tag=humble-tutorial) container can launch `moveit2_tutorials demo.launch.py` with or without the `rviz_tutorial` argument as well as a true or false remapping of it. Does your installation fail when the argument is left out entirely?

Alternatively, I suggest using `ros2 launch moveit2_tutorials demo.launch.py rviz_config:=panda_moveit_config_demo_empty.rviz` to bring up an empty rviz, along with following the instructions for [Saving your RVIZ Configuration](https://moveit.picknik.ai/main/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html#saving-your-configuration) so that you can customize RVIZ to your liking, save your changes, and use them in the future

hideto008, 2023-03-04:
I am facing same issue.

When I execute following command , rvis don't appear in screen.

```
$ ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true
```

When I execute following command, rvis appear in screen with Panda

```
$ ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=false
```
[image]

MikeWrock, 2023-03-04:
@hideto008 do you have any problems if you drop `rviz_tutorial:=` entirely and just use `ros2 launch moveit2_tutorials demo.launch.py`?

I also welcome you to try using `ros2 launch moveit2_tutorials demo.launch.py rviz_config:=panda_moveit_config_demo_empty.rviz` so you can save your preferred RVIZ config and reuse it

hideto008, 2023-03-05:
@MikeWrock 

Thanks.

I can't success to execute commands you suggested without error.

I tried `ros2 launch moveit2_tutorials demo.launch.py` and `ros2 launch moveit2_tutorials demo.launch.py rviz_config:=panda_moveit_config_demo_empty.rviz` and these command return same error about " `rviz_tutorial` does not  exit".

Is there any solution for this?

```
$ ros2 launch moveit2_tutorials demo.launch.py 

[INFO] [launch]: All log files can be found below /home/user/.ros/log/2023-03-05-08-42-38-903678-user-Alienware-X51-R2-6590
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): launch configuration 'rviz_tutorial' does not exist
```

```
$ ros2 launch moveit2_tutorials demo.launch.py rviz_config:=panda_moveit_config_demo_empty.rviz

[INFO] [launch]: All log files can be found below /home/user/.ros/log/2023-03-05-08-55-04-919807-user-Alienware-X51-R2-6830
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): launch configuration 'rviz_tutorial' does not exist
```

MikeWrock, 2023-03-05:
Thank you for trying that out for me, it would seem the `humble` branch still makes use of  `rviz_tutorial` and requires [these changes](https://github.com/ros-planning/moveit2_tutorials/pull/630) to appropriately set the defaults.

You will find the `main` branch has dropped the `rviz_tutorial` argument in favor of `rviz_config` used to specify the location of your desired RVIZ configuration, along with a number of other changes to [demo.launch.py](https://github.com/ros-planning/moveit2_tutorials/compare/main...humble#diff-a3efd55c9d8991e9a4a31bd3652364beaf91be99fd35307b50f8aa95b077a50c). I encourage you to take a look if you're interested in making a launch file do more in python.

closed 2023-03-21 by zflat